PX4/PX4-Autopilot #5807: Manual control lost message without RC connected

CarlOlsson, 2016-11-04:
Master fw 25be7aa7cf30b633afd1abeb30f15547719f1471
I calibrated airspeed without ever powering on the radio transmitter, constantly got "manual control lost" message in QGC, so maybe this is more a QGC issue

LorenzMeier, 2016-11-07:
Did you have a joystick connected?

CarlOlsson, 2016-11-11:
No I did not have a joystick connected

bigbellmercy, 2016-11-23:
Hi! I am undergoing the similar problem.
In fw v1.4.4 stable, the "manual control lost" happens also.
In fw v1.2 it does not happens when I tested.
It happens more frequently when XBEE modules are connected.
The RSSI value in QGC fluctuates under low value when it happens.

bigbellmercy, 2016-11-23:
And I suspect the RSSI value is correct because the Futaba RC transceiver send only RC IN signal to Pixhawk. It does not send RSSI info to Pixhawk.

julianoes, 2016-11-23:
Does it also happen when you set a parameter? RC lost tends to trigger if something takes a long time and the commander has a timeout (and therefore detects an RC loss because the timestamp of the last RC is far back).

bigbellmercy, 2016-11-24:
It happens all the time when I turn on RC controller irrespective of parameter changing.

julianoes, 2016-11-24:
@bigbellmercy your issue seems to be different from the one original issue reported here.
Please file a separate issue and describe it in detail.

bigbellmercy, 2016-11-24:
@julianoes You are right. I solved my problem. The cause was the signal noise in RC IN signal. By twisting the three wires (VCC, GND, RC_IN), it was fixed. The RC IN PWM signal seems to be sensitive to noise because its high data rate. Thanks~

julianoes, 2016-11-24:
@bigbellmercy thanks for the heads-up, glad it's resolved.

closed 2017-04-03 by mhkabir