PX4/PX4-Autopilot #5809: Is there Altitude+attitude OFFBOARD control?

mzahana, 2016-11-05:
Hi!

I would like to use a companion computer with px4 in order to control altitude, and attitude (roll, pitch in body frame). I know there is OFFBOARD mode, and I have been using it for position control. I also know it implements the following offboard mode:
* position setpoints
* velocity setpoints
* acceleration setpoints
* attitude(or attitude rates)+thrust setpoints

However, all of them are in local NED frame, and not in body frame. The attitude +thrust, however, are in body frame.

There is no mode that allows me to control altitude and roll/pitch(which is body frame). This is like altitude control mode (ALTCTL), where the user can hold altitude, and have free stick control on roll/pitch/yaw in body frame.

I though of using RC OVERRIDE, to switch to altitude control mode, and send rc signal from a companion computer to control roll/pitch/yaw. However, I am not sure if RC OVERRIDE is implemented in px4 firmware!

Is there a work around, or this is a new mode and has to be implemented ?

Thanks.

clovett, 2016-11-05:
I created something similar, see https://github.com/PX4/Firmware/pull/5774.  This gives you vx/vy velocity control with altitude position control.  It will pitch/roll as required to meet your vx/vy goals, while maintaining a fixed altitude.

mzahana, 2016-11-06:
Thanks, @clovett . That is interesting. However, I am wondering if it applies vx/vy in  a body-fixed frame, or local NED. What I am targeting is to command roll/pitch, or vx/vy in body fixed NED frame. In other words, if I command negative pitch (or +vx) it should move forward no matter what the heading is. Similarly, if I command positive roll (or +vy), it should move to right regardless of the heading.

Does your current implementation address the same issue?

mhkabir, 2016-11-06:
This implementation should take the setpoints in the local frame. However, projecting that to and from the body frame is very easy, so you can just add a stub which uses the current yaw to rotate frames.

mzahana, 2016-11-06:
@mhkabir that is right. Another solution I guess is to send a zero yaw setpoint, to make both frames aligned, right?

mhkabir, 2016-11-06:
Yes, you can do that. It would, however, mean that the quad would always point North. Which I'm not sure anyone would want to do.

mzahana, 2016-11-17:
I implemented a velocity controller to control altitude and velocities in body fixed frames (with simple projection from local to body frame using yaw angle). This is done in python+MAVROS. It works very well.

closed 2016-11-17 by mzahana

schmittlema, 2017-07-18:
@mzahana Could you post your code for this? It would be much appreciated. 
Are you sending yaw setpoints as well?

feilongfl, 2017-08-02:
@schmittlema you can ctrl yaw use code like this.

```

geometry_msgs::PoseStamped xyzyaw2Position(float x, float y, float z, float yaw)
{
    geometry_msgs::PoseStamped pose;

    pose.pose.position.x = x;
    pose.pose.position.y = y;
    pose.pose.position.z = z;

    pose.pose.orientation = tf::createQuaternionMsgFromYaw(yaw / 360 * 2 * 3.14);

    return pose;
}

```

use like:
```

ros::Publisher local_pos_pub = nh.advertise<geometry_msgs::PoseStamped>(
        "mavros/setpoint_position/local", 10);

local_pos_pub.publish(
                xyzyaw2Position(0, 0, 1.0, 0));

```

adipandas, 2018-09-22:
In this case how to send the velocity setpoint? Should I use the following topic: `/mavros/setpoint_velocity/cmd_vel`?

I am confused with the body frame and Local NED frame explanation provided [here](https://github.com/PX4/Firmware/issues/5809#issuecomment-258676420).

@mhkabir @mzahana By [this](https://github.com/PX4/Firmware/issues/5809#issuecomment-258676420) does it mean in gps/mocap denied environment, pixhawk takes the local_NED frame and it is aligned as the following:
N -- along positive-body-X axis
D -- along negative-body-Z axis
E -- along negative-body-Y axis

It will be very helpful, if someone can clarify this.
Thanks a lot for your help in advance.

HappyKidShiYu, 2019-07-31:
> 
> 
> I implemented a velocity controller to control altitude, and x/y velocities in body fixed frame (with simple projection from local to body frame using yaw angle). This is done using python+MAVROS on ODROID XU4. It works very well.

Hello! How did you implement your own contorller to maintain altitude?，Did you use thrust signal of mavros and deal with this using companion computer？Besides，have you tried with attitude？Thanks a lot~~~

HappyKidShiYu, 2019-07-31:
> 
> 
> I implemented a velocity controller to control altitude, and x/y velocities in body fixed frame (with simple projection from local to body frame using yaw angle). This is done using python+MAVROS on ODROID XU4. It works very well.
@mzahana
Hello! How did you implement your own contorller to maintain altitude?，Did you use thrust signal of mavros and deal with this using companion computer？Besides，have you tried with attitude？Thanks a lot~~~