PX4/PX4-Autopilot #5811: fixedwing_control demo crashes sitl

squilter, 2016-11-06:
I tried to run the fixedwing_control demo, but it would always crash with no error message. I started up sitl with `make posix gazebo_plane`, running Gazebo 7.4.0. When I started ex_fixedwing_demo, Gazebo immediately quit. 

```
pxh> ex_fixedwing_control start
Restoring terminal
[100%] Built target gazebo_plane
```

bkueng, 2016-11-08:
Looking at the source, this is quite old code, that hasn't been maintained and reworked for SITL Posix. The first step would be to replace the `exit(0)` calls with `return 0`, but there may be more steps necessary to make it work.
It does not crash, it just exits the whole `px4` process instead of only a single thread.

dagar, 2017-04-11:
@squilter were you able to fix this?

closed 2017-04-11 by dagar

squilter, 2017-04-12:
No, I was not.

dagar, 2017-04-13:
Do you still want to? The example is mainly there to show code usage, but is out of date for SITL usage.

squilter, 2017-04-13:
No thanks, you can keep it closed.