PX4/PX4-Autopilot #5824: What is the meaning of altitude in a LAND mission item for MC and VTOL

sanderux, 2016-11-08:
Currently the altitude parameter in a LAND command for multicopter or VTOL doesn't seem to have any function. 

When flying a mission: takeoff (10m) - waypoint (20m) - land (10m) 
the vehicle follows FOH navigation from the takeoff waypoint at 10m to the next waypoint at 20m. 
After that it proceeds at 20m to the land waypoint and lands at 0m. 

For FW it seems clear that the altitude is meant to indicate where to flare.
For MC and VTOL i would think the expected behavior is to consider the altitude parameter the same as an altitude parameter on a regular waypoint. Depending on settings the vehicle could follow FOH or ZOH towards the waypoint and begin the landing sequence at the given altitude.

@dagar @AndreasAntener @LorenzMeier any objections?

Ref: https://github.com/PX4/Firmware/pull/4785

dagar, 2016-11-08:
LAND contains the coordinates where you want to land, not where to begin landing. In FW it calculates the slope and altitude to flare at so that you hopefully hit the ground near the LAND waypoint.

It seems to me it should be possible as a user to configure only the last X meters of MC land to respect landing descent speed, but otherwise behave like a normal position setpoint (including FOH and descent speed).

AndreasAntener, 2016-11-08:
> For FW it seems clear that the altitude is meant to indicate where to flare.

It's not that simple, as @dagar stated.

For MC/VTOL I agree, the altitude should be respected first, then land from there.

dagar, 2016-11-08:
Isn't it ultimately the same thing if the user specifies the coordinates where to land or specifies the coordinates where to begin landing? Either way the implication is the user knows roughly where the ground is to setup a safe landing. The vast majority of MC flights are takeoff and land at home (relative alt 0).

Why change the semantics of MAV_CMD_NAV_TAKEOFF between FW and MC?

AndreasAntener, 2016-11-08:
If treated as ground altitude, can you explain how you imagine the MC to land in that case?

AndreasAntener, 2016-11-08:
@sanderux I think @dagar has a good solution. We discussed we will treat the altitude on LAND and VTOL_LAND as the position where you want to land (ground, default relative 0) and add a parameter to specify the approach altitude **relative to the landing waypoint altitude**. That way a multirotor and a VTOL can approach at a safe altitude (for instance default to 15m for MC, 30m for VTOL), descend at a *fast* landing speed, and then still use the final descent speed if set (as in #5828). This would comply with specs, work with a terrain model based approach and also with on board terrain estimation.

sanderux, 2016-11-08:
@AndreasAntener @dagar 
I see the usefulness of that specifically when the takeoff and land are not identical and there is a desire to use final approach speed. But i do also see once drawback; in this case there is no way to set the altitude of the FW waypoint that the VTOL_LAND command implies. If i fly at 150m but dont want to back transition at this altitude it would always need an intermediate waypoint. 

Also from a UX perspective it doesnt sound logical. The LAND and VTOL_LAND commands behave exactly as waypoints and there is no difference in the name 'Altitude'. I think that asking anyone without prior knowledge what behavior that would imply i think they would expect the behavior i suggested, mainly because landing at an altitude other than ground level is impossible. Also having the parameter set might be confusing when using a range finder as i would assume it would take precedence over the given value.

dagar, 2016-11-08:
I guess I just disagree that VTOL_LAND currently implies a FW altitude. The command only says "land using VTOL mode". It's the same way MAV_CMD_NAV_LAND works now for both FW and MC. The end user/operator enters the position (on the ground) where they'd like the vehicle to go, not the position in the air where they'd like landing to start.

LAND and VTOL_LAND behave the same way because of additional PX4 logic that arguably falls outside of the mavlink spec. As a user it's not unreasonable to think LAND would try and land in whatever VTOL state the vehicle is currently in, and VTOL_LAND means MAV_CMD_DO_VTOL_TRANSITION + MAV_CMD_NAV_LAND. VTOL_TRANSITION is a command that does not contain a position.

If we want a single VTOL_LAND waypoint to also specify the altitude to complete the transition at then I think it should be an additional optional param in MAV_CMD_NAV_VTOL_LAND. Doesn't this cover everybody's use case and keep LAND waypoint usage consistent across the board?

sanderux, 2016-11-08:
From a pure MC point of view, i even consider the regular LAND command's altitude implies "fly here at this altitude and then land'. But thats purely from a UX point of view. If the spec disagrees on logical behavior then i would rather change the spec then change the behavior to comply with a spec if that makes it do illogical stuff.

But maybe i'm wrong on what users expect to happen.

dagar, 2016-11-09:
That's not how LAND works for either MC or FW.

When starting MAV_CMD_NAV_LAND the current altitude is maintained until above the land, and then the z velocity is forced to MPC_LAND_SPEED. The description of MAV_CMD_NAV_LAND is "Land at location". If you look at MAV_CMD_NAV_LAND_LOCAL it's even more explicit. Instead of altitude you provide "Z-axis / ground level position [m]".

If you set your LAND @ 20m, and your last waypoint before it was at 30m it does this.
____________________    30m
                                        |
                                        |
                                        |
                                       X  <- LAND position setpoint @ 20m
                                        |
                                        |
                                        |
                                        |
                                       \|/
----------------------------------------------------------------------------

The autopilot uses the LAND position on the ground to plan an appropriate landing to hit that point. It lets the user tell the vehicle roughly where the ground is. If it has better information from a sensor or terrain data it can use that instead.
For a plane that means calculate an appropriate glide slope and flare to hit that point. For a multicopter instead of slowly descending at 0.5 m/s the entire way we can actually come down much faster until the last few meters.

sanderux, 2016-11-09:
@dagar i know it is not how it works now. My point was that i think it is how the user expects it would work. The question is if the way it works is how it should work.

dagar, 2016-11-16:
Internally QGC has simple and complex mission items. Complex mission items are multiple nav commands, but treated together as a block in QGC. 

So I think we'd have a complex item for land that would be final waypoint + land waypoint.
This is something I wanted to do in QGC anyway for a FW landing.

https://github.com/mavlink/qgroundcontrol/issues/3934

dagar, 2017-04-30:
I think we resolved this?

closed 2017-04-30 by dagar