PX4/PX4-Autopilot #5830: VTOL Takeoff weathervane

dagar, 2016-11-08:
Allowing VTOL to weathervane seems to be a great, but it can be abrupt at takeoff depending on the wind. I think we should only allow it with some altitude.

dagar, 2016-11-08:
@sanderux @AndreasAntener other than the first few seconds after takeoff and aligning to transition in the desired direction is there any reason we shouldn't have weathervane always on (optionally)? From what I've seen in logs it looks strictly better.

sanderux, 2016-11-08:
I have no objection to the param, but i would not like to enable it by default. A VTOL that hovers for any other reason than takeoff or land will likely be carrying a camera and heading is quite important for that.

VTOL_TAKEOFF and VTOL_LAND has a weathervane param already. That would only leave a regular transition command suffering from bad behavior after it completes transition and still has too much velocity. Perhaps a param that enables weatervane for the first waypoint after back transition would be an idea.

sanderux, 2016-11-08:
ref; https://github.com/PX4/Firmware/issues/5306#issuecomment-259250341

dagar, 2016-11-08:
Disagree on the any other reason part, but yes it would definitely be off by default.

Any thoughts on requiring some altitude first?

sanderux, 2016-11-08:
I dont see yawing directly after takeoff as a problem. Indeed it shows the user clearly where it will transition towards.

dagar, 2016-11-08:
The problem is if part of the vehicle is still touching the ground.

dagar, 2016-11-24:
Current situation seems fine with a bit of user education.

closed 2016-11-24 by dagar