PX4/PX4-Autopilot #5835: MavROS telem2 not able to arm and "Vehicle did not respond to request parameter" message.

pietrop88, 2016-11-10:
Hi all,

I run `roscore` on my pc and I have a pixhawk with px4 v1.4.4 connected to the pc through wifi and mavros. Mavros runs on a Edison board on the uav.
This is my launch script:

```
<?xml version="1.0"?>
<launch>
<arg name="ns" default="hexa3"/>
<arg name="fcu_url" default="/dev/ttyMFD1:921600" />
<arg name="gcs_url" default="udp://:14555@10.0.0.100:14550"/>
<arg name="system_id" default="103" />
<arg name="component_id" default="240" />
<arg name="tgt_system" default="103" />
<arg name="tgt_component" default="50" />
<node name="mavros" pkg="mavros" type="mavros_node" output="screen">
      <param name="fcu_url" value="$(arg fcu_url)" />
      <param name="gcs_url" value="$(arg gcs_url)" />
      <param name="system_id" value="$(arg system_id)" />
      <param name="component_id" value="$(arg component_id)" />
      <param name="target_system_id" value="$(arg tgt_system)" />
      <param name="target_component_id" value="$(arg tgt_component)" />
      <rosparam command="load" file="$(find hexa_test)/config/px4_pluginlists.yaml" />
      <rosparam command="load" file="$(find hexa_test)/config/px4_config.yaml" />
      <!-- enable heartbeat send and reduce timeout -->
      <param name="conn_heartbeat" value="5.0" />
      <param name="conn_timeout" value="10.0" />
      <param name="startup_px4_usb_quirk" value="true" />
      <param name="position/vision/pose_with_covariance" value="true" />
 </node>
</launch>
```
If I run it, I am able to see all the ROS messages that MavRos streams. This is the `/diagnostics` topic:
```
---
header: 
  seq: 10
  stamp: 
    secs: 946686152
    nsecs:   9621268
  frame_id: ''
status: 
  - 
    level: 0
    name: hexa3/mavros: FCU connection
    message: connected
    hardware_id: /dev/ttyMFD1:921600
    values: 
      - 
        key: Received packets:
        value: 2003
      - 
        key: Dropped packets:
        value: 0
      - 
        key: Buffer overruns:
        value: 0
      - 
        key: Parse errors:
        value: 0
      - 
        key: Rx sequence number:
        value: 37
      - 
        key: Tx sequence number:
        value: 92
      - 
        key: Rx total bytes:
        value: 81874
      - 
[... 480 lines cut ...]
        key: Timesyncs since startup
        value: 132
      - 
        key: Frequency (Hz)
        value: 9.826388
      - 
        key: Last dt (ms)
        value: -0.166981
      - 
        key: Mean dt (ms)
        value: 0.001059
      - 
        key: Last system time (s)
        value: 2498.510046000
      - 
        key: Time offset (s)
        value: 1478786426.792149782
```
The main difference is that now I have a **Time Sync**.

I think the main problem is in the connection between edison board and pixhawk through TELEM2.

Can you help me? 

P.

vooon, 2016-11-10:
Check electrical connection on **Edison Tx → TELEM2 Rx**. Try to sniff that line with logic analyser (or USB-UART Rx pin), check that Edison actually send data (connect Rx-Tx to test loopback).

Also date on Edison looks weird, install NTP daemon.
```
$ date -d @946686152 
Sat Jan  1 03:22:32 MSK 2000
```

vooon, 2016-11-10:
Note that `sys_time` (timesync driver) has magic timestamp *1234567890* (`2009-02-14T02:31:30+03:00`) to verify correct date is set.

pietrop88, 2016-11-11:
I have a problem with the connection **Edison Tx → TELEM2 Rx**.
Thank you very much.

LorenzMeier, 2016-12-26:
Stale / HW interface issue.

closed 2016-12-26 by LorenzMeier