PX4/PX4-Autopilot #5836: Pixhawk Mini -> No Sensor recognized

GabrielKoenig, 2016-11-10:
I have a problem with the new 3DR Pixhawk mini. Connecting it in QGC gives me the following message: "Parameters missing from firmware: -1:Cal_GYRO0_ID,-1:CAL_ACC0_ID,-1:CAL_GYRO0_ID,-1:CAL_GYRO0_ID: You may be running an older version fo firmware QGC does not work correctly with or your firmware has a bug in it." I have already updatet the newest QGC version and loaded the newest Firware of PX4 flight stack ond the Pixhawk mini. There ist no possibility to calibrate the sensors (Gyro, Acc, Compass). There comes always the message: No Mag Sensor 0, No ACCEL Sensor 0, No Gyro Sensor 0. I have also tried to set up the ID's manually (known from normal Pixhakws 3DR), but the only message it gave out, that no primary gyro was found. Thanks for help. 

[image]
[image]

GabrielKoenig, 2016-11-14:
Additional Informations: 
I have three Pixhawks Mini. Two of them show the same error described above. 
The third Pixhawk seem to work better. I can calibrate the Gyro and Accel, but no chance with the Compass. When I startet with the calibration of the compass first (all other sensors aren't yet calibrated), the following error message occurs: "ERROR: Retry Callibration (sphere NaN, 0), Critical [cal] calibration failed: mag".

Secondly I tried to calibrate the Gyro and Acc first, which worked in order to continue callibrating the compass. An other error message appears, compared to the previously shown: " Critical: attitude estimator not running -check system boo"

GabrielKoenig, 2016-11-23:
The issue can be closed, I had to load the developer firmware on the Pixhawk Minis.

closed 2016-11-23 by GabrielKoenig

tiralonghipol, 2018-03-15:
Hi ! I'm having the same issue..
I tried flashing the Developer firmware but still fail compass && level horizon calibration..
Any hint ?

LorenzMeier, 2018-03-15:
Try re-doing the gyro and accel calibration. If that doesn't help: Consider re-setting the board config by selecting the airframe config again. Is this the first time you are configuring this board?

tiralonghipol, 2018-03-15:
I've repeated the steps 'to the death':
- flashed different firmware versions (v2, v3, developer, v1.7beta)
- plug/unplug usb to reboot after every calibration
- change calibration order (gyro before/after compass)

nothing seems to fix this.. 
Everything was good with the pixhawk1

steveuk11, 2018-03-21:
Hi
Is there any solution to this?

I have 2 Pixhawk Minis and on neither can I get them to recognise the radio input with the latest stable firmware, on tested hardware used for other projects trying both SBUS and PPM. And both Minis giving me lots of errors, including the above. 

Can anyone recommend a firmware version that works with the Pixhawk Mini please?

dagar, 2018-03-21:
@steveuk11 have you tried PX4 master?

steveuk11, 2018-03-21:
@dagar just tried and still no luck

is there a way to reset everything on the Mini back to stock and start again? 

can provide any logs, pics, videos, screenshots etc to try and find a solution to get these to work. Thanks
[image]

tiralonghipol, 2018-07-09:
Hi. I know the Pixhawk mini is now discontinued, but is there any improvements on this?
any working combo of qgc and px4 version?