PX4/PX4-Autopilot #5837: SERVO_OUTPUT_RAW still has 0.9 format and causes error in jMavSim

sytelus, 2016-11-11:
It looks like Firmware is using definition of SERVO_OUTPUT_RAW message that is not updated to 1.0 specs. As per 1.0 specs there should be 16 servo fields so that payload size is 37. However PX4 sends message with only 8 servo fields so that payload size is only 21. This cause exception in jMavSim that gets silently ignored because jMavSim expects 37 byte payload size. It looked to me that Mavlink 0.9 had only 8 fields but 1.0 was upgraded to 16 fields. However may be the C header file never got updated?

dagar, 2016-11-24:
9-16 are extensions (mavlink 2)
https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L2611

lovettchris, 2016-11-30:
I added support for extensions in jMAVLib, see change [e14581e8c994690607a17450f36dd1f1dea818c1](https://github.com/PX4/jMAVlib/commit/e14581e8c994690607a17450f36dd1f1dea818c1).

sytelus, 2016-11-30:
closing this was fixed by lovittchris

closed 2016-11-30 by sytelus

kd0aij, 2016-11-30:
@lovettchris Do you need to update the jMAVlib submodule reference in jMAVSim?

lovettchris, 2016-11-30:
Yes, and I just submitted a push request on jMAVSim to do that.  You can do it yourself by doing this:
cd jMAVlib
git fetch
git checkout e14581e8c994690607a17450f36dd1f1dea818c1