PX4/PX4-Autopilot #5842: Preflight checking of navigation health is weak

priseborough, 2016-11-12:
Preflight checking does not make use of the EKF health data and dual sensors to detect sensor faults that would lead to loss of flight control after takeoff. The following should be added to PreflightCheck.cpp:

- Check of EKF horizontal and vertical velocity and position innovations.
- Check of EKF IMU bias estimates
- Where multiple IMU's are fitted, check of differences between accel and gyro readings
- Where multiple magnetometers are fitted, check of difference between magnetometer readings

priseborough, 2016-11-13:
I've created a prototype here: https://github.com/priseborough/Firmware/tree/pr-preFlightChecks

The next step is to enable pre-flight logging via parameter setting and enable adjustment of the thresholds via parameters.

I'm still trying to work out a way of accessing multiple instances of the IMU.

priseborough, 2016-11-13:
This will need to be reworked to move the processing into the sensors app because the preflight check is one-shot, but we need averaged/filtered values. I've made a start here: https://github.com/priseborough/Firmware/tree/sensorDataValidation-wip

LorenzMeier, 2016-11-13:
@bkueng Can you support this effort on the comms side? @dogmaphobic We will need your support for visuals in QGC.

LorenzMeier, 2016-11-13:
@DonLakeFlyer Let us know if you have any canned pre-flight check visuals in QGC already we could be using.

dogmaphobic, 2016-11-13:
I'll talk to Don and see about it. As of now, I have no idea where this would go.

AndreasAntener, 2016-11-13:
We also had a quick look into this recently, the SYS_STATUS would contain sensor flags based on the enum `MAV_SYS_STATUS_SENSOR` here: https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L399

But AFAIK they are never used/filled in PX4. Then I saw that Don already started pulling those in: https://github.com/mavlink/qgroundcontrol/pull/4206

Would those fields be sufficient to report or does it need more?

priseborough, 2016-11-13:
I've got it to a point where it is working on the bench (I can change sensor offsets and get it to fail) so have submitted a PR so we can get some review and discusion.

priseborough, 2016-11-14:
Creating a message that captures the data used for pre-flght checking would enable the same data to be both logged each time the check is run and also sent to the ground.

closed 2016-11-19 by LorenzMeier