PX4/PX4-Autopilot #5843: Protection against fly-aways due to navigation yaw errors is weak

priseborough, 2016-11-12:
An inflight check of velocity and magnetometer innovations is required to check for large yaw errors that at the moment will lead to a fly-away if a hovering type vehicle (eg multi-rotor) is left in a position or velocity control mode.

A check is required to put the vehicle into a non-GPS mode failsafe condition if this occurs.

stephanbro, 2017-01-20:
@priseborough I saw an interesting paper [here](www.mdpi.com/1424-8220/14/11/20008/pdf) where inclination can be used to validate the magnetometers health.

I've implemented something similar in the past using a lookup table, similar to the existing declination table, and comparing it with the current inclination. I could re-create an implementation if you think that could be useful.

priseborough, 2017-01-20:
It would be useful once the vehicle is in up and away flight, but on the ground and at low altitudes, there can be significant sources of external magnetic anomaly due to steel in building structures and vehicles.

stephanbro, 2017-01-20:
There are definitely issues with external magnetic fields from the things you mentioned, but also rebar in the ground as well. The previous implementation I mentioned worked well to deal with the low altitudes as well. The resolution of the inclination table would be the same as the declination table, 1 degree increments, so the GPS positioning accuracy can be quite low.

priseborough, 2017-07-18:
This has been addressed by https://github.com/PX4/Firmware/pull/6750 and https://github.com/PX4/ecl/pull/294

closed 2017-07-18 by priseborough