PX4/PX4-Autopilot #5854: How to check which Firmware release (version) is flashed on my Pixhawk FC

aitjellal, 2016-11-14:
Dear All,
I need to check which Firmware release (version) is flashed on my Pixhawk FC.
- I can access to Pixhawk using both : Qgroundcontrol and MAVROS
- no NettShell
best regards

julianoes, 2016-11-14:
In QGC you should open the mavlink inspector widget and look out for http://mavlink.org/messages/common#AUTOPILOT_VERSION or if you can access mavlink shell using Tools/mavlink_shell.py (instead of QGC), then you can use the command `ver all`.

aitjellal, 2016-11-14:
Thank you @julianoes ,
I still could not solve the problem:
1 -  Using python script:
I have downloaded the Firmware source and  compiled using CatKin then
when I try to start the python script:  it does not connect. This is what I get. (my pixhawk Baud is set to 921600)
`$ python mavlink_shell.py -b 921600 /dev/serial/by-id/usb-FTDI_FT232R_USB_UART_AL01OWID-if00-port0`
` Using MAVLink 1.0`            
`Connecting to MAVLINK...`

it stays like this. 

2 - Using QgroundControl: 
I could not find "AUTOPILOT_VERSION " on the mavlink inspector widget. Here is a screen-shot of my QGC (version 2.7.1)
[image]

LorenzMeier, 2016-11-14:
Your QGC is hopelessly outdated and so is your PX4 version. Upgrade to PX4 1.4.4 or later and QGC 3.1 or later. The autopilot version field will work then. You can also run `ver all` on the shell.

closed 2016-11-14 by LorenzMeier

eduardkieser, 2021-03-14:
Hi I also need to find out the fw version flashed to my Pixhawk 4, but I suspect that things my have changed somewaht since this solution was posted because I dont see an AUTOPILOT_VERSION message in the mavlink inspector of QGC.

HadesGuy, 2022-11-17:
Same here. 
Using version  QGC 4.2.4 
Please give some insight. 
Thanks

julianoes, 2022-11-18:
Please don't necrobump. Create a new issue instead and provide details.