PX4/PX4-Autopilot #5855: FMUv1 hangs on boot

julianoes, 2016-11-14:
FMUv1 fails booting with master. This was reported in http://discuss.px4.io/t/mpu-sensor-problem/1667.

`git bisect` points to c24046ca5bcfc220b579ab5229af8fbf5010816c
```
c24046ca5bcfc220b579ab5229af8fbf5010816c is the first bad commit
commit c24046ca5bcfc220b579ab5229af8fbf5010816c
Author: Lorenz Meier <lorenz@px4.io>
Date:   Sun Sep 4 20:55:50 2016 +0200

    FMUv1: Save resources in config
```

LorenzMeier, 2016-11-14:
I reviewed the diff and compared to other configs - we came up with final numbers close to the tested original. I've undone the commit, fixed in 0ca397ab2046c2a2a71b3ca758e1cbb6f80a87fb.

julianoes, 2016-11-14:
I wouldn't go as far as undoing the whole commit especially because it negatively affects tap.

I would go with just #5856.

PR #5856 (2016-11-15): px4fmu-v1: Raise INTERRUPTSTACK again
This fixes #5855, tested on FMUv1.
commit 25d1daa7 (2016-11-15): px4fmu-v1: Raise INTERRUPTSTACK again

This fixes #5855.
closed 2016-11-15 by LorenzMeier