PX4/PX4-Autopilot #5857: compass calibration error : out of memory.

haitanyise, 2016-11-15:
[image]
 firmware: v1.5.0  hardware: PX4FMU_V4.

pavel-kirienko, 2016-11-15:
As a quick workaround, you could disable dynamic node ID allocation by setting `UAVCAN_ENABLE` to 1, and assign the node ID of your GNSS receiver manually.

LorenzMeier, 2016-11-16:
@haitanyise Try again with current master, not the 1.5 RC.

haitanyise, 2016-11-18:
I tryed the latest master branch today and  found new problems.
1. can't download onboard sd card file.

haitanyise, 2016-11-18:
I tryed the latest master branch today and  found new problems.
1.Can't download onboard sd card file.
2.Uavcan cannot allocate dynamic ID.
3.Vtol airframe sensors can't calibration properly.
[image]

LorenzMeier, 2016-12-26:
We optimised RAM and it should be fine now.

closed 2016-12-26 by LorenzMeier