PX4/PX4-Autopilot #5862: How much of the firmware code is tested?

mkschreder, 2016-11-15:
How much of your codebase is it that is currently being fully covered by unit tests? (besides some libraries like matrix)

dagar, 2016-11-16:
Hard to say. There are some tests here (https://github.com/PX4/Firmware/tree/master/src/systemcmds/tests) and here (https://github.com/PX4/Firmware/tree/master/unittests) that run in the builds every pull request, but the coverage is probably fairly low. We also have mission tests that run every build. They use multicopter, fixedwing, and vtol controllers and cover a huge amount of the core codebase.

Are you interested in adding more testing?

mkschreder, 2016-11-16:
> Firmware/tree/master/src/systemcmds/tests) and here (

Interesting with the mission tests. It should not be very hard to enable
gcov/lcov coverage reporting on all tests that run during build. However I
doubt that mission tests test extreme case behaviours for various
components so even if they may cover code they may not exhaustively cover
data range scenarios.

I'm obviously interested in adding unit tests to all components but for px4
it would be a tremendous task to add after the fact (it would require me to
understand in detail every single component and probably refactor a lot of
things to make unit testing of individual blocks easier). Such things
should not be implemented by one person but instead by everyone who commits
changes to the project.

I think that from quality assurance point of view 100% code coverage would
be a huge achievement. Even better if it is done on several levels (ie some
tests test each component individually and others test higher level
components while at the same time also testing lower level components as
well). Once unit testing is in place you can enable valgrind on unit tests
and hunt down all kinds of obscure programming errors that may sometimes
creep in.

dagar, 2016-11-17:
It would be great to establish a baseline with gcov/lcov. Let me know if you're interested in adding it to the builds. We have additional capacity on semaphoreci right now.

I think a lot of the difficulty is knowing how to test something. We have an okay test framework, every commit builds, but it oftens seems quite difficult to add testing.

One interesting thing with PX4 is that every module (https://github.com/PX4/Firmware/tree/master/src/modules) is really just uorb messages in, uorb messages out. Our new logger records these messages, and we can selectively replay them. What do you think about using this mechanism to bootstrap more exhaustive testing? We could collect data from real flights that test a wide range of features and selectively replay comparing the real world and simulator output. We could use code coverage metrics to guide the specific test flights.

mkschreder, 2016-11-17:
How come "make check" builds arm targets? Shouldn't it only build native? Also, what is the correct place to add LINK_FLAGS or COMPILE_FLAGS but only to all modules when building test/sitl target? I don't think adding it in px4_add_module would be entirely correct because I only want to add certain flags for a native build. Any ideas?

davids5, 2016-11-17:
make check is for CI build testing, @dagar where are these defined?

dagar, 2016-11-17:
How about in here? https://github.com/PX4/Firmware/blob/master/cmake/posix/px4_impl_posix.cmake#L179

The idea with `make check` is to build everything we care about and make it easy for developers to match CI. `make tests` will be posix_sitl_default and run the testing, but I haven't added mission tests to it yet.

mkschreder, 2016-11-17:
Interesting.. so apparently UNIX is not LINUX. :-) I had to add my definitions in the else block instead. 

compiler: -fprofile-arcs -ftest-coverage --coverage 
linker: -ftest-coverage --coverage -lgcov

    make tests

    lcov --directory . --capture --output-file coverage.info
    lcov --remove coverage.info '/usr/*' --output-file coverage.info
    lcov --list coverage.info
    genhtml coverage.info --output-directory coverage-html
    chrome-browser coverage-html/index.html

Results: 12% coverage from ~46000 lines of code that is being tested.

dagar, 2016-11-17:
Nice! Well, obviously not the coverage level, but at least we can start measuring.
Can you try running the mission tests? 
http://sitl01.dronetest.io/job/PX4%20Firmware%20POSIX%20SITL%20PRs/2943/

Can we use a service like coveralls, or should we just dump the html on s3?

mkschreder, 2016-11-17:
Yes you can use coveralls.

Travis file for coveralls upload:

```
before_install:
  - pip install --user cpp-coveralls
  - gem install coveralls-lcov
after_success:
  - cd ${TRAVIS_BUILD_DIR}
  - lcov --directory . --capture --output-file coverage.info # capture coverage info
  - lcov --remove coverage.info '/usr/*' --output-file coverage.info # filter out system and test code
  - lcov --list coverage.info # debug before upload
  - coveralls-lcov coverage.info # uploads to coveralls
```

mkschreder, 2016-11-17:
@dagar how do I run the mission tests?

dagar, 2016-11-18:
http://dev.px4.io/tutorial-integration-testing.html

It'll be a bit more complicated because we use containers, but we can definitely get this in the builds right away. We'll probably need to do it on semaphore instead of travis-ci due to capacity.

Can you open a PR to add the flags to the build behind some option to enable? Instead of putting the details in CI config can it be in the top level Makefile? Everything needed to build posix_sitl_default with the right options, start the capture, and run the tests, and generate the output all behind a single target? Something like `make tests_coverage`.

dagar, 2016-11-24:
I'll look at adding coverage to the build on semaphore.

@mkschreder Should we have separate jobs for running the tests with and without coverage on?

mkschreder, 2016-11-24:
No I think all tests should use coverage flags. Coverage drives unit
testing.

dagar, 2016-11-24:
Sorry that wasn't clear, I meant in the build system where we run testing.

Can the testing always run with the coverage flags on, or should we have a regular testing build, and a testing + coverage build.

mkschreder, 2016-11-25:
I really don't see why it wouldn't. Any particular reason you want to have
two separate jobs? I would just enable everything on the main test job.
Testing can always run with coverage on. Then if for whatever reason you
can not afford tests running a bit slower, you can set up a different job
without coverage..

closed 2016-12-07 by LorenzMeier

Alberto070, 2017-10-03:
Hi all,
I'm trying to use gcov on the px4 target but having problems when linking. 
<<In function `gcov_version': libgcov.c:(.text+0xac): undefined reference to `_impure_ptr'>>
Anyone faced similar issue ? 
Thanks in advance.

Used the following configuration:
My gcc version 4.6.2, 
EXTRACFLAGS = -fprofile-arcs -ftest-coverage --coverage 
LDFLAGS += -ftest-coverage --coverage -lgcov

mkschreder, 2017-10-03:
Generate a map file using -Wl,-Map,output.map and perhaps there you can see
what references that symbol. Likely some kind of libc/newlib issue.

mkschreder, 2017-10-03:
If whatever is referencing it is something that only gets called upon
exiting the application you can probably do a temporary fix by just
defining a pointer called impure_ptr somewhere and the problem will go away
since your px4 application never exits. But you should also try a different
toolchain just to see if it would provide a proper fix.

Alberto070, 2017-10-03:
Hi  @mkschreder 
Thanks so much for your answer.
Unfortunately the reference is in many places (not only gcov_exit) so a fake _impure_ptr pointer wouldn't do the trick.
As you correctly pointed out it is an issue coming from newlib.  _impure_ptr was added to make "errno" management reentrant.  
What Ido not understand is how is it possible that  gcc libs have NOT impure_ptr while the gcov lib (coming whitin the same toolchain) make use of them.
Any ideas ? 
thanks,
A

Here follow the complete linker log.
c:/work/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/armv7e-m/fpu/libgcov.a(_gcov.o): In function `gcov_version':
libgcov.c:(.text+0xac): undefined reference to `_impure_ptr'
c:/work/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/armv7e-m/fpu/libgcov.a(_gcov.o): In function `__gcov_open':
libgcov.c:(.text+0x2b4): undefined reference to `setbuf'
c:/work/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/armv7e-m/fpu/libgcov.a(_gcov.o): In function `buffer_fn_data':
libgcov.c:(.text+0x524): undefined reference to `_impure_ptr'
c:/work/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/armv7e-m/fpu/libgcov.a(_gcov.o): In function `gcov_exit':
libgcov.c:(.text+0x810): undefined reference to `_impure_ptr'
libgcov.c:(.text+0x814): undefined reference to `_impure_ptr'
libgcov.c:(.text+0x856): undefined reference to `_impure_ptr'
libgcov.c:(.text+0x85a): undefined reference to `_impure_ptr'
c:/work/px4/toolchain/bin/../lib/gcc/arm-none-eabi/4.7.4/armv7e-m/fpu/libgcov.a(_gcov.o):libgcov.c:(.text+0x886): more undefined references to `_impure_ptr' follow
make[1]: *** [/c/WORK/px4/devel/Firmware/Build/p-fcsw.build/firmware.elf] Error 1
make[1]: Leaving directory `/c/WORK/px4/devel/Firmware/Build/p-fcsw.build'

dagar, 2017-10-03:
PX4 has a gcov configuration. Try running cmake with `CMAKE_BUILD_TYPE=Coverage`.

Alberto070, 2017-10-03:
Hi dagar 
thanks for your answer.
Our software is based on a previous (old) Firmaware version and unfortunately cmake configuration was not implemented at that time. 
In any case I used exactly the same compier options used by "CMAKE_BUILD_TYPE=Coverage" and the result was exactly the same.

Can you please tell me which gcc version are you using ? 
Thanks in advance,
A

mkschreder, 2017-10-04:
Maybe a question you should be asking is why you are using old version of
px4 AND an ancient version of gcc. Latest version of gcc is gcc 6.

As for ideas on why gcov lib uses it and not other libs I think it may be
that gcc was built incorrectly. gcc build is a two step process where you
first build a minimal version of gcc, then build newlib using that and then
build a final version of gcc with newlib baked into it. It is likely that
libgcov could have accidentally been linked against wrong libc. But instead
of trying to figure that out I would just grab a latest binary release of
arm gcc cross toolchain for your linux distro and try that first instead.

Alberto070, 2017-10-04:
Hi mkschreder,
thanks for your answer.
Your first question is a very good question I'm also asking myself for at least the last year.  The  short answer is that (unfortunately) px4 version to use is not under my control.
Since gcc is under my control I downloaded the lastest arm-none-eabi distro (for windows) and I'm trying to use  in the old toolchain. 
Unfortunately I've hit another bug and I'm still digging into it. 
<<c:\work\px4\toolchain\arm-none-eabi\include\c++\6.3.1\bits\basic_string.h:5443:31: error: 'strtoll' is not a member of 'std'
   { return __gnu_cxx::__stoa(&std::strtoll, "stoll", __str.c_str(),>>

Thanks very much for your explanation about the gcc building process.

mkschreder, 2017-10-04:
To use strtoll you must use -std=c++11 flag when compiling. It is only
available in c++11.

Alberto070, 2017-10-04:
Yes, I saw that and added:
ARCHCXXFLAGS     = -std=c++11 
to the toolchain configuration.
Unfortunately it is not enought.

Thanks a lot Martin.
A.

PR #5969 (2016-12-07): add tests code coverage
 - closes #5862 

Run with `make tests_coverage` and then open coverage-html/index.html in a browser. I'll be adding this to the builds and post output to https://coveralls.io/ soon, and then try to bring in the mission tests.

@mkschreder please review
commit 1941dfca (2016-12-07): add tests code coverage

-closes #5862