PX4/PX4-Autopilot #5867: RAM usage on VTOL

eyeam3, 2016-11-15:
I ran into some issues regarding RAM usage when using a VTOL airframe in combination with UAVCAN. If I switch to a generic quadrotor as airframe, I did not have these issues.

Subsequently, a comparison of memory usage on VTOL and multirotor when the UAVCAN is not activated.
Generic VTOL:
```
             total       used       free    largest
Mem:        235872     208096      27776      23872
```
Generic quadrotor:
```
             total       used       free    largest
Mem:        235872     189968      45904      45728
```

mhkabir, 2016-11-15:
Related : https://github.com/PX4/Firmware/issues/5857

dagar, 2016-11-24:
@eyeam3 did you run out of memory on VTOL with uavcan? Sorry if this is obvious, but VTOL is always going to use more memory than MC, as it's also running the FW controllers and the VTOL controller.

dagar, 2017-04-28:
We're actively watching this now. 
https://github.com/PX4/Firmware/issues/7146

closed 2017-04-28 by dagar