PX4/PX4-Autopilot #5870: Implement groundspeed_undershoot for flying without airspeed sensor

sanderux, 2016-11-16:
Currently groundspeed undershoot changes the target airspeed
https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1015

When flying based on throttle setting this would be bypassed and risk fly-aways

dagar, 2016-11-17:
Isn't TECS only using potential energy with airspeed disabled?

dagar, 2017-01-02:
@sanderux any thoughts on how you'd like to implement this?

sanderux, 2017-01-03:
Not yet, do you have any idea's?

dagar, 2017-01-04:
Just an idea that it's not an obvious addition :wink: 

Perhaps a part of TECS in no airspeed mode to maintain a minimum groundspeed.

sanderux, 2017-01-04:
why not obvious? there is a groundspeed_undershoot detection for vehicles with an airspeed sensor, so wh not for vehicles without one?

dagar, 2017-01-04:
Not obvious how to implement. As you know groundspeed undershoot just adjusts airspeed. The rest of the controller is oblivious.

sanderux, 2017-01-05:
Yes, the way to scale it might be tricky. perhaps it should add some throttle, wait to seewhat the ground speed does, add a little more, etc

LorenzMeier, 2017-04-10:
Is this done since then?

sanderux, 2017-04-10:
Not yet

bresch, 2017-04-18:
" > perhaps it should add some throttle, wait to seewhat the ground speed does, add a little more, etc "

@sanderux Something like a simple integrator controller?

closed 2018-02-21 by PX4BuildBot