PX4/PX4-Autopilot #5881: Can not download sd card file

haitanyise, 2016-11-19:
I tryed the latest master branch yesterday and found new problems.
1.Can't download onboard sd card file.
2.Uavcan cannot allocate dynamic ID.
3.Vtol airframe sensors can't calibration properly.
[image]

dagar, 2016-11-24:
1. Can you try the daily build of QGC. It has new log download functionality.
 2. @LorenzMeier @pavel-kirienko Any ideas for this? 

    > Uavcan cannot allocate dynamic ID.

3. Can you provide more information? Any error messages?

pavel-kirienko, 2016-11-24:
> Uavcan cannot allocate dynamic ID.

I'm not sure what exactly does this mean. Dynamic node ID allocation server is failing to start or what?

haitanyise, 2016-11-26:
I set  UAVCAN_ENABLE :Dynamic ID/Update, and when pixhawk start, there is no allocation of fixed note ID.
[image]
[image]
[image]

haitanyise, 2016-11-26:
[image]

haitanyise, 2016-11-26:
@pavel-kirienko

pavel-kirienko, 2016-12-01:
I'm not sure what behavior you expected to see. There is no such thing as
allocation of fixed node ID - "fixed" means that allocation is not needed.
PX4 always uses node ID of 1 for itself.

closed 2016-12-26 by LorenzMeier