PX4/PX4-Autopilot #5894: tune PID parameter

applededipan, 2016-11-21:
i have a six rotor aircraft on hand, the weight is about 10Kg, the wheelbase is about 1680MM, the default pid parameter dose not work well, so can anybody give me some advice about tuning the pid parameter?

AlexisTM, 2016-11-22:
What do you mean by not working? 

What is the problem? What is the multicopter behavior? Is it too aggressive or drunk copter?

ardente, 2016-11-27:
哥们。你是哪的呀？你的问法太中式化了 😁

julianoes, 2016-11-27:
It's always a good idea to share a log and/or video. Otherwise it is very hard to give advice.

(And please stick to English here, thanks).

applededipan, 2016-11-28:
i see @julianoes

applededipan, 2016-11-28:
it didn't work well, not "not working", it was  too aggressive, but now it's better. Thanks!!!

applededipan, 2016-11-28:
it didn't work well, not "not working", it was too aggressive, but now it's better. Thank you!!! @AlexisTM

applededipan, 2016-11-28:
shandong @ardente

AlexisTM, 2016-11-28:
@applededipan The log download function (new in last QGroundControl version) allow you to download the log (.px4log file). You upload the log to [Log Muncher](http://logs.uaventure.com/) and it will plot relevant data!

[image]

To tune the PID, you can go in the "tuning section" of qGroundControl. NOTE : Correctly read the informations. I would suggest putting `Roll` & `Pitch Sensitivity` to the right, `Attitude Control Sensitivity` to the left and `Position Control Sensitivity` to the left.

AlexisTM, 2016-11-29:
@applededipan What are your current parameters (Multicopter Attitude Control) ?

LorenzMeier, 2016-12-26:
Closing as stale.

closed 2016-12-26 by LorenzMeier