PX4/PX4-Autopilot #5895: Documentation needed for estimator tuning for high vibration airframes

julianoes, 2016-11-22:
In the [weekly dev call](http://discuss.px4.io/t/weekly-dev-call-of-november-21-2016/1770) it was suggested that we document how the estimators can be tuned for airframes with high vibrations.

@jgoppert and @priseborough, is this already somewhere in the Devguide? Could you shed some light on this?

This would resolve #5453.

priseborough, 2016-11-22:
There is currently no tuning guide for the ecl EKF (we do need one, which means I need to write it). The best technique with ecl is to gather a replay log and tune off-line using ekf2_replay. If vibration is causing estimation errors due to aliasing, then gyro and accel process noise parameters can be increased. The innovation levels are the best guide to overall estimator performance.

LorenzMeier, 2016-11-28:
@priseborough Can you prioritise the documentation for that? It starts to gate the release and I think its really important at this point. It should go into Gitbook. You might need to sign up (log in with Github, don't sign up with one of the other services): https://www.gitbook.com/book/px4/devguide/edit

priseborough, 2016-12-01:
OK, i will start on it tomorrow.

closed 2016-12-02 by LorenzMeier