PX4/PX4-Autopilot #5911: Gimbal controller plugin throwing OGRE_EXCEPTION

L4ncelot, 2016-11-26:
Hi,

I've been playing with gimbal and it's controller plugin in **Gazebo**. I'm using the same model of camera and gimbal as in **typhoon_h480** model. 

Everything works fine until I add to the model `librotors_gazebo_gimbal_controller_plugin` with this code:
```
<plugin name='gimbal_controller' filename='librotors_gazebo_gimbal_controller_plugin.so'>
            <joint_yaw>iris::cgo3_vertical_arm_joint</joint_yaw>
            <joint_roll>iris::cgo3_horizontal_arm_joint</joint_roll>
            <joint_pitch>iris::cgo3_camera_joint</joint_pitch>
            <imu>camera_imu</imu>
</plugin>
```

Then the **Gazebo** throws this message:
```
terminate called after throwing an instance of 'Ogre::ItemIdentityException'
  what():  OGRE EXCEPTION(5:ItemIdentityException): Can not find codec for 'png' image format.
There are no formats supported (no codecs registered). in Codec::getCodec at /build/buildd/ogre-1.9-1.9.0+dfsg1/OgreMain/src/OgreCodec.cpp (line 69)
```

I'm using fresh installation of **Gazebo 7.4** and latest version of **px4** firmware.

Has anyone idea what's going on here? Thank you for any advice.

bkueng, 2016-11-28:
Hi
Looks like you're missing some `png` library on your system. Seems strange though that it fails on `gimbal_controller`, as it does not use images.
Which OS & version do you use? Did you follow the dev-guide installation?
Can you test with gdb to see from where the exception is thrown?

Do you have the same problem with the typhoon_h480 config?

L4ncelot, 2016-11-28:
Hi @bkueng, yeah it is very strange since I think it worked for me previously. The error is thrown even when I use typhoon_h480 model with `make posix gazebo_typhoon_h480` command. I'm running it on Ubuntu version 14.04.

L4ncelot, 2016-11-30:
So after some struggle with graphic card drivers I ended up reinstalling my current OS and installing **Ubuntu 16.04**. I wanted to do this a long time ago, because of some graphical and other issues with my old OS (it was just broken) anyway...

Now I no longer have any issues with **OGRE** exception and everything works fine.

closed 2016-11-30 by L4ncelot

julianoes, 2016-11-30:
Cool, thanks for the feedback.