PX4/PX4-Autopilot #5913: Lidar-lite (PWM) not being detected in 1.5.1

SDroneS, 2016-11-26:
Hi guys, 
I've just began my adventure with PX4 firmware developing and from the very beginning I've got some troubles. I downloaded the latest 1.5.1 version of the firmware and connected PXflow and LIDAR-lite (via PWM) to my Pixhawk. I set parameters to use local_position_estimator and LIDAR (SENS_EN_LL40LS->enabled).
When I connect to the nsh console all I can see is PXflow's sonar detected with ID=0 but there is no sign of lidar detection. When I type "pwm_input test" the test is carried out as it should be what means, as I understand correctly, that the LIDAR is connected properly and its driver had to be previously started.
When I change firmware from 1.5.1 to 1.4.4 (via QGC) everything works perfectly! LIDAR is detected with ID=0 and PXflow's sonar with ID=1. I tried to analyze differences in code and to me all what's related to distance sensors detection remained the same as in 1.4.4. What's wrong???

BTW I think that I found mistake in local position estimator in 1.5.1. In `local_position_estimator/sensors/lidar.cpp` account for leaning is doubled in lidarMeasure() and in lidarCorrect(). Check that.

Thanks for any help in advance.

julianoes, 2016-11-27:
The driver is currently not compiled for Pixhawk because flash space is low.
Try to build the source yourself and enable the driver here:
https://github.com/PX4/Firmware/blob/master/cmake/configs/nuttx_px4fmu-v2_default.cmake#L29
Maybe, you need to disable something else instead that you're not planning on using.

SDroneS, 2016-11-27:
It works, thanks a lot!
One more question - is it better to stop PXflow sonar advertising when connected both LIDAR and PXflow? In previous version they worked together.

julianoes, 2016-11-27:
I think the laser range sensor should take priority over the sonar from PX4Flow, otherwise something is not implemented the right way. The data from the laser has shown much more useful and less noisy than sonar.

SDroneS, 2016-11-27:
Probably there's something wrong with data fusion in EKF. Right now we got rid of redundancy. 
There's no doubt that laser is better than sonar but if you use PXFlow you have sonar on-board anyways. Even if you have less accurate sensor on-board there's no point in neglecting it's measurements.

julianoes, 2016-11-27:
The sonar data from PX4Flow that I've seen is not really useful. You have to reject a lot of the data to get something useful.

Of course, you can prove me wrong by creating a pull request that uses both distance sensors and works better :).

SDroneS, 2016-11-28:
I'll work on that when I have some time. Thanks for help :)

julianoes, 2016-11-28:
@SDroneS No worries.

We need to add the information about building the ll40ls driver into the Devguide. I'll keep the issue open until that's done.

dagar, 2018-01-17:
I believe this is resolved? Reopen if I'm wrong.

closed 2018-01-17 by dagar