PX4/PX4-Autopilot #5914: Fixedwing with ulog logger: Inspect CPU load

LorenzMeier, 2016-11-28:
@bkueng The CPU load here is high and I'm wondering if this is logger related:
http://logs.px4.io/plot_app?log=bb728903-b655-4ea4-b650-635f25c3147b

bkueng, 2016-11-28:
The logging rate it is quite high, compared to my bench test (~84KB/s vs 51KB/s), which comes from topics like `position_setpoint_triplet`, `rc_channels` and `input_rc`.
A local test increasing the logging rate to ~87KB/s does not reveal increased CPU usage:
```
 PID COMMAND                   CPU(ms) CPU(%)  USED/STACK PRIO(BASE) STATE
   0 Idle Task                   16686 43.546     0/    0   0 (  0)  READY
   1 hpwork                        781  2.340   820/ 1592 192 (192)  w:sig
   2 lpwork                        118  0.354   364/ 1592  50 ( 50)  READY
   3 init                         1406  0.000  1132/ 2496 100 (100)  w:sem
 295 top                           449  3.191  1236/ 1696 100 (100)  RUN
 100 gps                           125  0.567   908/ 1192 220 (220)  w:sem
 102 dataman                         1  0.000   772/ 1192  90 ( 90)  w:sem
 136 sensors                      1027  2.907   988/ 1696 250 (250)  w:sem
 138 commander                    1655  3.971  2772/ 3192 140 (140)  w:sig
 139 commander_low_prio              1  0.000   556/ 2992  50 ( 50)  w:sem
 148 mavlink_if0                  2325  7.588  1724/ 2392 100 (100)  READY
 149 mavlink_rcv_if0                 2  0.000   956/ 2096 175 (175)  w:sem
 153 mavlink_if1                   489  1.489  1716/ 2392 100 (100)  w:sig
 154 mavlink_rcv_if1                 2  0.000   956/ 2096 175 (175)  w:sem
 169 mavlink_if2                   381  1.063  1636/ 2392 100 (100)  w:sig
 172 mavlink_rcv_if2                 2  0.000   956/ 2096 175 (175)  w:sem
 190 frsky_telemetry                 0  0.000   508/ 1096 200 (200)  w:sem
 221 log_writer_file               505  1.843   572/ 1016  60 ( 60)  w:sem
 208 mavlink_if3                  1226  3.546  1644/ 2392 100 (100)  w:sig
 209 mavlink_rcv_if3               725  2.269   956/ 2096 175 (175)  w:sig
 219 logger                       2053  6.595  3036/ 3592 250 (250)  w:sem
 246 ekf2                         5008 16.028  4972/ 5792 250 (250)  w:sem
 248 fw_att_control                534  1.985   904/ 1392 250 (250)  w:sem
 255 fw_pos_ctrl_l1                 27  0.000   764/ 1696 250 (250)  w:sem
 270 navigator                       7  0.000   932/ 1296 105 (105)  w:sem
```

So I don't think it's the logger. Needs further investigations.

closed 2016-12-02 by LorenzMeier

julianoes, 2016-12-02:
The triplet should only be published when it changes like for multirotors, so not very often at all.

bkueng, 2016-12-02:
That's not what I'm seeing in MC SITL mission flight though:
```
TOPIC NAME                        INST #SUB #MSG #LOST #QSIZE
position_setpoint_triplet            0    5   83   144 1
```
So around 80Hz. And from the provided log file I get a similar rate.

julianoes, 2016-12-02:
@bkueng ah I didn't realize that it's published more often during a mission.

AndreasAntener, 2016-12-03:
@julianoes @bkueng the triplet is constantly updated during missions for FOH and heading control.

bkueng, 2016-12-05:
@AndreasAntener Ah ok, thanks. As long as it's intended to update at this rate, it's all good.