PX4/PX4-Autopilot #5920: 'Advanced' parameter meta flag

sanderux, 2016-11-28:
Add a flag to indicate if a parameter is considered 'Advanced'
UI can then hide them to make regular parameters more accessible

dagar, 2016-11-29:
If this was discussed on call I missed it.

As a first pass which would you consider not advanced? https://pixhawk.org/firmware/parameters

A much lower/more obvious bar might be to first hide all the params that aren't intended for direct user editing.

 - CAL_*
 - RC
 - SYS_AUTOCONFIG, RC_DSM_BIND

sanderux, 2016-11-29:
It was discussed at the start of the dev call with @LorenzMeier 
i think in essence all parameters are advanced, but some of them are super advanced.

The following parameters i would consider not very advanced and would like to make them less scary and less buried between the super advanced params;
- Everything that has a setting window in QGC
- Circuit brakers, people that need to set them will find them anyway
- Weathervane and other VTOL tuning options
- PID gains, at least the ones that require attention
- Pitch and roll offsets
- Allow arming without GPS
- EKF2/LPE IMU, OF and GPS positions
- FW L1 limits, options and tuning params
- TECS airspeed, throttle, climb, sink and tuning params
- Follow target params
- Gimball
- Land detector MAX values
- Launch detection tuning
- LPE home pos
- MAVLINK ID's
- Mission tuning (1st waypoint max dist, min takeoff alt, altmode, yaw timeout, etc)
- MC max values
- MPC tuning, min and max values
- PWM reversal, min and max
- RX type
- Roll / pitch trims
- RTL tuning
- Runway takeoff tuning
- SDLOG UTC offset
- IMU / Compass rotation
- Rangefinder enablers
- UART ESC setttings
- LED brightness
- Estimator selector
- TELEM2 baud rate
- Logger
- Failsafe channel mapping

sanderux, 2016-11-29:
I think param meta data should get a level indicator like 
```
@level [Advanced or Simple]
```
And default to Advanced

dagar, 2016-11-30:
Interesting, that's way more than I expected. What's your definition of simple vs advanced? If circuit breakers and estimators aren't advanced I don't know what is. If we start with everything unlabelled and defaulting to advanced we could slowly bring the others forward only as needed.

I like this idea, I'm just wondering if there's an opportunity here to have an additional level that strips it down to the level appropriate for an end user. Imagine if you configured and tuned a particular vehicle ready to handoff. Is there a level that would be appropriate for them? Think DJI only allowing you to configure RTL altitude and a handful of other things. Thinking out loud...

@DonLakeFlyer what do you think?

DonLakeFlyer, 2016-11-30:
I kind of consider the Parameter Editor as a whole a power user feature. Hence everything in there is for advanced users. The simple is covered by QGC ui. That also does not mean there is missing QGC setup screens that should be done as well. That said it's easy enough to filter on an advanced bit. I think you may want to make whole groups advanced though not individual parameters.

LorenzMeier, 2016-11-30:
@DonLakeFlyer Would you be able to work with our users then to get these setup screens here?  @MaEtUgR can you propose the parameters (the ones mapped to a slider) for a position control and an attitude control tuning screen?

MaEtUgR, 2016-11-30:
Most important rate control parameters in my opinion are (these are best/only tunable in acro mode):
MC_ROLLRATE_P, MC_PITCHRATE_P, set MC_ROLLRATE_D, MC_PITCHRATE_D, MC_YAWRATE_P
Attitude control parameters are:
MC_ROLL_P, MC_PITCH_P, MC_YAW_P

Note that for a nearly symmetric multicopter roll and pitch values should not have to differ for decent performance, yaw values certainly do.
I consider these Parameters advanced and they have to be handled with care because setting them too low renders the vehicle hard to control and setting them to high can lead to very unsafe instant flip (Better have them too low than too high).
For the position controller I'm not yet able to judge what is important.

sanderux, 2016-11-30:
I think there might be some confusion due to the wording 'Simple' and 'Advanced'
I think only power users should enter the parameter screen, but at the moment there are many parameters that are too much even for power users and they are starting to outnumber the parameters that are required for tuning.

Perhaps calling the levels 'Expert' and 'Advanced' would be better. And maybe in the future a 'Basic' level could be added for very simple params. The advantage of having this filter is not only usability, it also points power users to params that could be interesting for tuning their setup. 

@DonLakeFlyer would it be possible to hide a group when there are no params that meet the filter?

DonLakeFlyer, 2016-11-30:
> would it be possible to hide a group when there are no params that meet the filter?

Yes. But I'd be careful to create this whole new feature just to make tuning easier. A full tuning page is scheduled for 3.2.

DonLakeFlyer, 2016-11-30:
Any suggestions for tuning page can go here: https://github.com/mavlink/qgroundcontrol/issues/3911

sanderux, 2016-11-30:
> Yes. But I'd be careful to create this whole new feature just to make tuning easier. A full tuning page is scheduled for 3.2.

The aim is not to only allow tuning, it is to make parameters that could require attention more accessible. Like mission behavior or logger choices. It's not just tuning.

hamishwillee, 2017-07-24:
I'm not sure about simple  vs advanced, but it would be good to hide "not relevant" parameters. So for example there are a whole bunch of FW parameters that don't need to be shown for MC and visa versa (e.g. loiter radius)

dagar, 2017-07-24:
@hamishwillee most of that we would already get for free if we're more careful to respect module boundaries or to only find and copy those params when the system is configured as the relevant type.

closed 2018-02-21 by PX4BuildBot