PX4/PX4-Autopilot #5924: Complex mixer not loading with more than 3 inputs

Tiktiki, 2016-11-30:
It seem there is a problem with the mixer module.
When ever you try to use more than 3 inputs in any mixer, it does not load.
Theses inputs can be from any control group, the issues still is present.

If there is a mixer that does not load, the next mixers in the file also won't load.
It does not throw any errors in the console however.
The loaded mixers along with every other function of the flight controller seem to work correctly.

Tests have been done on a hardware bench test using a Pixfalcon/Pixhawk mini and digital AND analog servos.

Example mix that does not work:

M: 4
O: 10000 10000 0 -10000 10000
S: 1 0 2250 2250 0 -10000 10000
S: 1 1 -3750 -3750 0 -10000 10000
S: 0 0 3000 3000 0 -10000 10000
S: 0 1 -5000 -5000 0 -10000 10000

Removing ANY input from the mix makes it work.

dagar, 2016-12-09:
Sounds like it's not a px4io problem (same issue on px4fmu). I quickly tried a mixer like this on a pixhawk and didn't see any error messages.

AndreasAntener, 2016-12-10:
There are certain combinations where the mixer send/parse flow doesn't work, maybe related: https://github.com/PX4/Firmware/issues/5118
Parsing on the receiving side fails and stops silently.

LorenzMeier, 2016-12-28:
@Tiktiki Can you retry on current master? I fixed a race there.

LorenzMeier, 2016-12-28:
From Andreas:

> If we try to load the following mixer it seems to get sent to IO but one of the last 2 mixers doesn't get loaded.
> 
> ```
> R: 2- 10000 10000 10000 0
> Z:
> M: 2
> O:      10000  10000      0 -10000  10000
> S: 1 0  10000  10000      0 -10000  10000
> S: 1 1  10000  10000      0 -10000  10000
> M: 2
> O:       10000   10000    0 -10000  10000
> S: 1 0   10000   10000    0 -10000  10000
> S: 1 1  -10000  -10000    0 -10000  10000
> ```
> 
> It seems to split it up at a bad location during transfer. If I extend it a bit (add another null mixer before), or remove one of the minus signs in the first servo mixer, or change `PX4IO_P_CONFIG_MAX_TRANSFER` from 64 to 32, the mixer loads correctly.
>

LorenzMeier, 2017-01-01:
Please test this. It should fix it.

AndreasAntener, 2017-01-04:
@Tiktiki this should be fixed on master now

closed 2017-01-04 by AndreasAntener

Siytek, 2017-08-19:
Hi all, I'm still getting this problem with the latest stable release, which I re-flashed today just to be sure. The mixer fails to load every time I boot up and also fails to load manually from the nsh prompt.

```
Addons script: /fs/microsd/etc/extras.txt 
WARN  [px4io] fmu sent: "M: 2
O: 10000 10000 0 -10000 10000
S: 0 0 -8000 -8000 0 -10 "
px4io mixer send error -1
ERROR [mixer] failed to load mixers from /fs/microsd/etc/mixers/X8.main.mix
ERROR [mixer] failed to load mixer
```

The mixer is just a copy of the X5 mixer with added pass-thru for the other main PWMs. I had been using it with previous firmware with no trouble for quite a long time.

```
# Skywalker X8 mixer for PX4FMU

# Output 1 & 2 Elevon mixers

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -8000  -8000      0 -10000  10000
S: 0 1   6000   6000      0 -10000  10000

M: 2
O:      10000  10000      0 -10000  10000
S: 0 0  -8000  -8000      0 -10000  10000
S: 0 1  -6000  -6000      0 -10000  10000

# Output 3 Null

Z:

# Output 4 Motor speed mixer

M: 1
O:      10000  10000      0 -10000  10000
S: 0 3      0  20000 -10000 -10000  10000

# Output 5 to 8 pass-thru

M: 1
O:      10000  10000      0 -10000  10000
S: 3 4  10000  10000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 3 5  10000  10000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 3 6  10000  10000      0 -10000  10000

M: 1
O:      10000  10000      0 -10000  10000
S: 3 7  10000  10000      0 -10000  10000
```

Siytek, 2017-08-19:
I have a bad habit of scratching my head trying to solve a problem for hours and then as soon as I post about the issue I realise I'm doing something stupid :) Flashed the latest _master_ firmware and the mixer is loading ok!

LorenzMeier, 2017-08-20:
So master is fine, correct?