PX4/PX4-Autopilot #5927: Fixed Wing Offboard Ignores Attitude Setpoint

aivian, 2016-11-30:
The fixed wing controller ignores attitude setpoints in offboard mode. I originally filed this over on mavros, but later found the problem in the px4 firmware (https://github.com/mavlink/mavros/issues/630)

I've tentatively traced it to the fixed-wing attitude controller which generates attitude targets from the manual inputs whenever climb rate control is disabled (line 959, https://github.com/PX4/Firmware/blob/master/src/modules/fw_att_control/fw_att_control_main.cpp)

adding another condition for offboard mode fixes it, but I'm not familiar enough with the controller structure to know if that is likely to introduces problems elsewhere.

Original:
`if (!_vcontrol_mode.flag_control_climb_rate_enabled) {`

With additional condition:
`if (!_vcontrol_mode.flag_control_climb_rate_enabled && !_vcontrol_mode.flag_control_offboard_enabled) {`

dagar, 2016-12-11:
@fiji3119 FYI

dagar, 2016-12-11:
That's part of it, but you'll also want to prevent fw_pos_ctrl_l1 from publishing like MC does here - https://github.com/PX4/Firmware/blob/master/src/modules/mc_pos_control/mc_pos_control_main.cpp#L2201

aivian, 2016-12-16:
Ah, thanks. It did work after only the first few changes, weird. I guess the two setpoints were fighting for the controller's attention.

PR #6112 (2017-02-16): fixed wing offboard attitude setpoint fix
Fix #5927 where attitude setpoints from offboard were being overwritten by the RC channels due to some missing flags.
closed 2017-02-16 by dagar