PX4/PX4-Autopilot #5929: FTP commands can cause memory corruption

lovettchris, 2016-12-01:
sending FILE_TRANSFER_PROTOCOL message with opcode kCmdListDirectory with the payload->data set to a non-existant directory "/foo" hangs the Pixhawk in mallinfo().  If I comment out the _mavlink->send_statustext_critical calls in mavlink_ftp.cpp, then no problem.  So perhaps the generation of those error messages is overflowing the stack or corrupting the heap somehow.

dagar, 2016-12-01:
This might be a good excuse to restore the mavlink_tests. https://github.com/PX4/Firmware/tree/master/src/modules/mavlink/mavlink_tests

LorenzMeier, 2016-12-26:
@lovettchris Does this still persist?

lovettchris, 2017-04-04:
Yes.  If you build (AirSim)[https://github.com/Microsoft/AirSim], and this command:
    MavLinkTest -serial:/dev/ttyACM0,115200

then create a local empty file called "foo" and enter this command:
    put foo /fs/microsd/foo/bar

then px4 stops responding to all mavlink commands.

TSC21, 2017-08-20:
@julianoes did you get the chance to take a look at this?

julianoes, 2017-08-21:
@TSC21 sorry no chance, this is not something I'm familar with.

TSC21, 2017-08-21:
> @TSC21 sorry no chance, this is not something I'm familar with.

Sorry I wanted to ping @bkueng instead.

bkueng, 2017-08-21:
Closing, all symptoms indicate stack overflow/memory corruptions, which I fixed in https://github.com/PX4/Firmware/pull/7665

closed 2017-08-21 by bkueng