PX4/PX4-Autopilot #5931: Ground Truth based Estimator Benchmark

jgoppert, 2016-12-01:
Now that we have ground truth working in SITL I think it will be good to run each estimator through a mission and compute the errors for all of the states using GPS /vision/ and flow measurements. This has two benefits:

1. Unit testing to make sure performance isn't negatively impacted by a change
2. Unbiased comparison between estimators. This will be useful for consideration of switching the default multirotor estimator to EKF2 for v2.0. 

@LorenzMeier @bkueng

jgoppert, 2016-12-02:
Plots here of ekf2 vs lpe/ attitude_estimator_q

https://github.com/dronecrew/px4tools/blob/master/examples/ulog%20analysis.ipynb

This is almost complete, we just need to decide on some heuristic way to score the estimators based on their error variances and give a simple percentage that CI needs to meet and then we can also compare the two ratings between the estimators. Currently it looks like ekf2 has superior attitude and velocity peformance, but altitude is better with lpe. Also need some way of evaluating cpu load in SITL, not sure if that is possible, @LorenzMeier @AndreasAntener @bkueng .

priseborough, 2016-12-02:
Point 1) I agree with - there is great benefit to this for reversion testing
Point 2) I don't fully  agree with. SITL at the moment does not have the fidelity in sensor error and environment models to be able to make accuracy comparisons between algorithms or to optimise tuning.  However  SITL will be very useful to verify the response to specific failure cases, eg loss of GPS, misaligned magnetometer, ground based magnetic anomaly, etc that are impractical to verify through testing. I think it is the off-design and failure cases where we should concentrate the effort with in SITL.

ecmnet, 2016-12-02:
@jgoppert James, I just have a basic question: What is the source of the ground-truth in SITL? And could you give some hints how to reproduce your example?

jgoppert, 2016-12-02:
@ecmnet The source of the ground truth is the gazebo simulator. To reproduce this:

Need to have gazebo installed.

```bash
make posix_sitl_default gazebo
```

Then start qgroundcontrol and start a mission. The log will be in:

```bash
build_posix_sitl_default/tmp/rootfs/fs/microsd/log/2016-12-02/08_11_30.ulg  08_12_15.ulg
```

Instal px4tools

```bash
git clone git@github.com:dronecrew/px4tools.git
cd px4tools
python setup.py install
```

Now using python:

```python
import px4tools
data= px4tools.read_ulog('./your_path_to.ulg').resample('1L').concat()
data.estimator_analysis()
```

jgoppert, 2016-12-03:
@priseborough 

I do want to start handling the fault cases and trying to increase our gazebo sensor model fidelity. I realize there is a limit to what simulated testing can tell us, but the further we push it I think we will get a lot of benefit. It is always hard to sync with testers or have enough time to test every commit ourselves.

Handling sensor faults will be interesting. It would be great to have the ability to script failures etc in gazebo, something like:

mission:
1. takeoff
2. goto waypoint 1
3. goto waypoint 2
4. land

failtures:
@10 seconds, fail gps 1 for 5 seconds
@20 seconds, fail gyro 1
@30 seconds, manual control loss for 120 seconds

Then we would need some desired trajectory to compare against.

PR #6161 (2016-12-26): Adds ground truth tests to sitl gazebo CI.
Closes #5931.
closed 2016-12-26 by LorenzMeier