PX4/PX4-Autopilot #5942: conflicting input MAVLINK and RC receiver

cla-ftero, 2016-12-02:
When the px4 is connected with an RC receiver and "virtual joystick" enabled on the qgroundcontrol, we got fluctuating motor speed. When disabling it, the problem was solved. Maybe one could detect that a RC Receiver is connected and take either the input from it OR the input send by qgroundcontrol. Or show an error message.

dagar, 2016-12-02:
It would certainly be nice if we could both connected, but how would the autopilot decide which one to use and make that clear to the operator?

cla-ftero, 2016-12-02:
Couldn't the priority be set to either by default. and changeable with a parameter/setting?

LorenzMeier, 2016-12-02:
@dagar Since we have now `orb_publish_auto()` we could publish both topics, one with higher and one with lower prio and let commander choose based on prio.

dagar, 2016-12-02:
Great! Figuring out how and where to implement the priority was my only question.

cla-ftero, 2016-12-02:
I think this would be great since for us (new users) it wasnt trivial to find the problem.
Thanks for the fast answers.

dagar, 2016-12-02:
Ok, then the next issue is that it could be confusing or even dangerous if we lost RC (the priority) in a manual mode, but the failsafe doesn't trigger because it's still receiving input from QGC thumbsticks, and the user wasn't aware or ready.

We could try and be clever so that a loss of the active input still triggers the failsafe, but then make it possible for the secondary input to show valid activity. Like retaking control after a geofence violation and wiggling the sticks.

Initially I think the easiest/safest way to go is having a param like this
 1. RC Only
 2. Mavlink Only
 3. RC + Mavlink (RC Priority)
 4. RC + Mavlink (Mavlink Priority)

closed 2018-02-21 by PX4BuildBot