PX4/PX4-Autopilot #5945: Altitude estimation error of LPE. In OFFBOARD control with VICON.

Youngdong-Clark-Son, 2016-12-03:
Hi.

I am using the most updated firmware and new to PX4.

I am trying to fly multi-rotor with VICON.

I am sending vicon data with 100hz and the difference between LPE and VICON is negligible.

I checked it and flew a vehicle with setpoint_position and the result is as following.

[image]

The flight performance was bad due to the big error of 'z'. The position error between LPE and VICON is negligible as I said. But the velocity error is weird especially in 'z'. The velocity plot of VICON was plotted just using the Euler method of VICON data (e. g.  v(k) = (z(k+1)-z(k))/dt ).

The outdoor flight performance was much greater when I did position hold. Something is wrong.

Any idea how to increase the flight performance?

Thanks.

Additionally, attitude data of VICON and mavros.
[image]

jgoppert, 2016-12-03:
It seems that LPE is tracking vicon altitude. Are you sure you are sending the vicon altitude in the correct coordinate frame (NED). It appears that vicon altitude is +Z (which would be beneath the ground).

If this isn't the issue, and you want LPE to track the euler approximation for the velocity more, you could increase the process noise (PN_V). This is the process noise for the equation that d/dt(V) = A. This will make the state more uncertain and the kalman corrections will have more impact.

When changing these values, I would try to carry the vehicle around manually before trying a test flight.

Youngdong-Clark-Son, 2016-12-04:
Thank @jgoppert for the reply.

Can I ask one more question?

You said the correct coordinate frame of the Vicon is 'NED'. But as you can see the graph of mine, when I sent positive altitude setpoint, the vehicle took off and landed with negative altitude setpoint. Moreover, as shown in [http://dev.px4.io/ros-mavros-offboard.html](url), this setpoint of altitude is also positive for take off. This page said that 

> Even though the px4 flight stack operates in the aerospace NED coordinate frame, mavros translates these coordinates to the standard ENU frame and vice-versa.

Then do I need to send the Vicon data with 'NED' format? I think the problem of now is not for the frame, but for the wrong velocity estimation.

Anyway, I will try to increase the process noise of LPE and post results. Thank you!

Youngdong-Clark-Son, 2016-12-05:
Thanks to @jgoppert , I succeeded to fly with OFFBOARD. The flight result is as shown. To make FCU follow the velocities of Vicon, I increased the process noise(PN_V) of velocity propagation so that FCU heavily depends on the exact Vicon data.
[image]
[image]
[image]

  Imu data and Vicon data for attitude are almost same. The position data shows that the flight was very successful though some fluctuation still occurs in 'z'. I think this can be reduced by modulating the position controller gains.

  Still, I wonder why LPE makes messages 'reinit P (0, 0) negative' when I increase PN_V more than 0.6. I am currently using the value of 0.6. But this also makes the same message 'reinit P (0, 0) negative' at first. So I need to reduce it to about 0.4 and increase again to remove warning messages.

  The recommended range of PN_V is from 0(minimum) to 1.0(maximum). As the increased value makes the velocity estimation more accurate, I guess it is better to make PN_V near 1.0.

Thank @jgoppert for the amazing support.

closed 2016-12-07 by Youngdong-Clark-Son

weiweikong, 2016-12-19:
@Youngdong-Clark-Son @jgoppert  I saw your experiments with VICON and LPE is quite well. 

- Have you ever encountered the situation as shown in following figure

 
[image]

As the blue box shown, there is a strange overshoot at the very beginning when I swithed the UAV to OFFBOARD mode. Then the UAV maneuver back to normal.

- Have you ever 
- How did you send the VICION data to PX4? by `mocap` or `vision_pose'
- Did you publish the `set_position` all the time before switching to OFFBOARD mode?

The detail log was uploaded at http://logs.uaventure.com/view/PTMjRcANNg6y5fex4SU8Aj.

abdullahmohiuddin, 2017-06-04:
Hi @Clark-Youngdong-Son , I was having similar problems specially in X-axis, so I did the same thing as you did, I plotted the euler velocity that I obtained from Position estimation that I get from LPE ( from px4logs). There are some differences. So, if I do what you did with PN_V , I would be relying on Mocap position or LPE position for euler velocity estimation?