PX4/PX4-Autopilot #5947: Estimator performance eval

LorenzMeier, 2016-12-03:
I'm looking forward to having all of our effort focused in one direction. Here are some things about LPE that I would like to see incorporated:

Things to merge from LPE:

Chi-squared fault detection with beta lookup table so faults can be set based on false-alarm rate.
Measurements are modular and separated by source file.
To add a state or measurement to LPE once extends the state/measurement enum, the rest is handled automatically to resize the matrices and the jacobians are automatically updated since they are set sparsely by the state enum. This allows anyone to modify the estimator. If we are going to rely on a symbolic jacobian calculation, I would like to see it done in something like sympy so it can be released publicly for use by people without a matlab license.
Some things from EKF2 that I'm looking forward to

Replay logging support.
Attitude state covariance (since LPE uses attitude_estimator q).
More developers on the same page.
Some concerns about EKF2 that I would like to see addressed.

CPU load.
Transition from local navigation (using optical flow/ vision etc) to GPS navigation.
Does the cross covariance from the attitude states actually help with the state estimation. I would like to see a comparison because we are paying a lot to do that computation. Maybe a separate state/ attitude ekf is still better.
API bloat. We currently have to translate all calls from the px4 param server to the ECL api, see notes about ROS2 below.
Long term plans

What is the long term goal for the project? I think eventually we will all be using ROS to do the estimation in some form when ROS2 is working. I think we should align with this to the best of our ability with this new combined estimator. I would like to see it very closely resemble a ros node with params accessed through a clean param interface etc. EKF2's goal was to use ECL to separate it from the environment, but I think it should be written for ROS2 in the future, so all of the ECL interface will get in the way. Our current px4 param interface is actually closer to ros than what ECL is doing.

LorenzMeier, 2017-04-10:
Important, but not critical.

TSC21, 2017-07-25:
I think a way of complementing this should be adding the possibility of users write and use their own estimators - https://github.com/mavlink/mavlink/pull/667.

closed 2017-10-04 by LorenzMeier