PX4/PX4-Autopilot #5954: Send CI logs to Flight Review

jgoppert, 2016-12-03:
@dagar Suggested this and I wanted to write it down. We should have sitl_gazebo CI upload the logs from testing to logs.px4.io and provide links in the log build or some other way to tie them to the CI instance.

LorenzMeier, 2016-12-03:
@bkueng Can you take care of this after the Intel Aero polishing this week? Ideally the URLs would be sent by Jenkins as comments on the PR from PX4 build bot.

dagar, 2016-12-03:
We could start with failures only if capacity is a concern.

bkueng, 2016-12-05:
Sounds good.
@dagar where do I need to add this? Or do you want to do this?
Basically what we need:
- Send a POST request to logs.px4.io/upload, adding a tag `source` with value `CI`
- the response will be a redirect to the plots
- make sure not to follow the redirect, just get the link
- only upload on CI failure (we're currently somewhat size restricted on the server)

dagar, 2016-12-05:
Somewhere in here https://github.com/PX4/Firmware/tree/master/integrationtests, but @AndreasAntener might have a better idea.

Do a simple script for now? Could also be useful for developers.

    ./Tools/log_post.py ./build_posix_sitl_default/tmp/rootfs/fs/microsd/log/2016-12-05/19_06_47.ulg

Default to using git username and email if they weren't supplied. I'll keep dreaming...

AndreasAntener, 2016-12-05:
If the script is in src and takes a log path as argument we can easily call it from Jenkins.

dagar, 2016-12-05:
@AndreasAntener what about just the failures? Is an existing method for calling the mission tests individually? I've been thinking about breaking everything out (each individual unit test, mission) and reporting the results to something like cdash, circleci, jenkins junit plugin, etc.

AndreasAntener, 2016-12-24:
Done

closed 2016-12-24 by AndreasAntener