PX4/PX4-Autopilot #5955: Intel Aero tuning

LorenzMeier, 2016-12-03:
@bkueng @MaEtUgR Please run another tuning session on beta / this tag:

https://github.com/PX4/Firmware/releases/tag/v1.5.1rc3

Main performance issues:

- Yaw rate / position tracking is not great
- Roll / pitch tracking could still be a bit better
- Position hold (MPC_POS XY and VEL gains)
- Altitude hold (MPC_POS_Z and VEL_Z gains)

LorenzMeier, 2016-12-03:
@simonegu FYI - this should fit somehow into Monday if possible.

LorenzMeier, 2016-12-03:
@lucasdemarchi @lfrodrigues FYI.

lfrodrigues, 2016-12-03:
@LorenzMeier ?

MaEtUgR, 2016-12-03:
ok, it's on top of the list

bkueng, 2016-12-05:
@MaEtUgR @julianoes log files of the test flights are here:
acro + manual: http://logs.px4.io/plot_app?log=6699704e-c84b-454a-8e0c-eb0b199c4e30
manual + position: http://logs.px4.io/plot_app?log=60252e13-2a75-4a18-aa2c-c6babe43d0ae

In general the altitude hold was not good, this is something that needs to be looked at

MaEtUgR, 2016-12-05:
@LorenzMeier We went out tuning, yaw tracking rate and attitude improved (wasn't tuned last time), roll and pitch tracking was already tuned (gain on maximum value but we increased the rate limit). The position controller does work normally well on the Aero with the biggest problem in my opinion being altitude sags.

While fiddling around with the pos controller to get a better feel for tuning in sitl I found some unexpected behaviour... I'll see if #5971 already fixed it or else it's worth an issue.

MaEtUgR, 2016-12-07:
The unexpected behaviour I found will be fixed by #5980 .

lucasdemarchi, 2016-12-12:
> Altitude hold (MPC_POS_Z and VEL_Z gains

Is there a PR with that tuning that I missed?

LorenzMeier, 2016-12-12:
Reopening until that is in. @MaEtUgR ?

MaEtUgR, 2016-12-14:
No, there wasn't any change on this gains yet.
If it's not sufficient I and @bkueng maybe need to do another testflight.

LorenzMeier, 2016-12-14:
As discussed I find the flight performance still underwhelming - that was before @bkueng mounted the tube on the autopilot's pressure line.

LorenzMeier, 2016-12-14:
And the system load issues will also have contributed to the degradation:
https://github.com/PX4/Firmware/issues/6037

LorenzMeier, 2017-04-30:
Done

closed 2017-04-30 by LorenzMeier