PX4/PX4-Autopilot #5956: Intel Aero version check & auto-update

LorenzMeier, 2016-12-03:
@lfelipe @lucasdemarchi @dogmaphobic We need a way to do an automatic version check in QGC when connected to QGC and advise users to update. Otherwise we risk Aero users sticking to stale flight control firmware versions with suboptimal performance or missing critical fixes.

dogmaphobic, 2016-12-03:
@DonLakeFlyer already did that. The code that receives and parses the version is here:
https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/Vehicle.cc#L518-L539

The UI tells you you are using an old version of the firmware when you connect a vehicle. It currently tells you the version should be 1.4.1 or greater. 

What I noticed is that I get weird numbers coming from the firmware. For instance, right now when I connect the version I receive is 2559, which when parsed using the above code becomes 0.0.9.

LorenzMeier, 2016-12-03:
@dogmaphobic That's absolutely not what we need. What we need is something entirely different:

  * Get the software version on the vehicle
  * Get the latest software version online
  * Warn the user if the vehicle is older than online

This is nowhere but is urgently needed. The 1.4.1 check is just QGC's own compatibility and a one-time hack.

The version number you are getting is a valid development version. Please see the code in QGC decoding it. We don't care about developers, we care about released versions.

dogmaphobic, 2016-12-03:
I have no idea all that Don has done about this. He's added code to receive and parse the version from the vehicle. The *absolutely not what we need* is not really absolute.

What Don has added is part of the request in this issue. Now that QGC has the version info from the vehicle, it needs to know how to check for updates properly. That's the missing part. I don't know where the 1.4.1 comes from, if it's hardcoded or checked against something. This *something* is what needs to be defined. Where would this online query come from? I would also assume the version is vehicle dependent (implied by the title if this issue)? If so, the URL for the query would have to depend on what vehicle is connected?

How will QGC get this value is the question. Once that's established, it's just a matter of getting the value and using it for the compare. In theory, everything else is in place (assuming the piece identifying the vehicle is part of this same message Don is currently parsing).

I would rather hear from Don before doing anything as he would know a lot better what is there and what isn't.

LorenzMeier, 2016-12-03:
The mechanism for that should be based off this discussion:
https://github.com/mavlink/qgroundcontrol/issues/2797

Please ping @dagar for the PX4 side integration. He can refer to other devs if its too much workload.

LorenzMeier, 2017-10-05:
We need this in general for companion computers.

closed 2018-02-21 by PX4BuildBot