PX4/PX4-Autopilot #5959: Pixhawk 2.1 Support

dagar, 2016-12-03:
@kd0aij @nkhoit FYI

We actually don't seem to have Pixhawk 2.1 support at the moment. We need to start 2 instances of the mpu9250, another ms5611, and check the rotations. At the moment the external l3gd20 and lsm303d are failing to start in init, but work if I manually start them after. If I had to guess the IMU heater needs some type of basic support at a minimum. Sitting on my desk the sensors are currently > 70C.

Internal
mpu9250 -R 14 start
ms5611 start

External
mpu9250 -X -R 14 start
l3gd20 -X -R 4 start
lsm303d -X -R 6 start
ms5611 -X (?)

@LorenzMeier any idea how we keep this straight from the Solo? mpu6000 vs mpu9250?

proficnc, 2016-12-03:
For the board temperature, 60 degrees is target. All the drivers exist at the Ardupilot PX4 layer.

dagar, 2016-12-03:
It's closer than I originally thought. I missed pulling in the mpu9250 check that sets BOARD_FMUV3 true. I'll still verify everything.

davids5, 2016-12-03:
Can we use the board name?

dagar, 2016-12-04:
Pixhawk 2.1 was actually only missing the 2nd ms5611. Pixhawk 2 (solo) init was broken and falling back to regular px4fmu-v2.

PR https://github.com/PX4/Firmware/pull/5962

Pixhawk 2.1 imu heater is still a problem. The sensors were in the mid 70s after bench testing. I'll open a separate issue specifically for that.

closed 2016-12-04 by dagar