PX4/PX4-Autopilot #5963: Pixhawk 2.1 IMU Heater Support

dagar, 2016-12-04:
With current PX4 master the sensors reach exceed 70s during bench testing.

closed 2016-12-05 by LorenzMeier

kd0aij, 2016-12-05:
how was this resolved?

dagar, 2016-12-06:
@kd0aij Disabled on pixhawk 2.1 here - https://github.com/PX4/Firmware/pull/5964
Interested in adding a simple controller to maintain sensor temperature?

keenanjohnson, 2017-12-28:
I'm a bit confused about the state of this. Is the heater currently hard on or disabled for pixhawk 2.1 cubes?

LorenzMeier, 2017-12-28:
Completely off. The unit heats up by itself fairly quickly.

LorenzMeier, 2017-12-28:
It was added as an experiment by the designer. We are not using one in designs that are in mass-manufactured drones. We do use temperature calibration through extensively.

proficnc, 2017-12-28:
The heater is used to set the IMU’s to their optimal temperature for operation.

This is used within other flight controller stacks very successfully. 

When flying in extreme cold temperatures, it allows for setting an optimal temperature for operating, regardless of any additional temperature compensation.

This was not an experimental feature, this was a requirement of design. And though I respect that PX4 has chosen to go for compensation tables, gyro noise cannot be compensated for outside the recommendations of the manufacture... therefore I would still recommend that the heater is used by PX4 to set a minimum viable temperature, and also, as recommended by Paul R, a method for doing a full range temperature compensation setup.

Of course, this is 100% up to the PX4 team.

jzazbert, 2019-04-08:
@dagar 
We have discovered that the heater is not 100% off as of 1.8.2 firmware. PX4 sets the io pin to be OPEN_DRAIN, and then sets it high, which leaves the heater control pin floating. Because of interactions with the resistor network and the transistor that powers the heaters, the heat is on a small amount, that gets worse as temperature increases.

By disabling the green "RING" LEDs and changing the drive pin for the heater to PUSH_PULL and setting to high, we get the heater to turn completely off. Both that it was partially on and that this fix turned it all the way off were experimentally measured on 2.1 hardware.

With lights off and the heater completely off, we've seen 20-25C drop in sensor temperatures inside the cube. We had to make this change because we were pushing sensors out of spec at only 45-50C ambient. We can now calibrate to almost 60C ambient without pushing sensors past absolute operating max, and still have a little headroom. The MPU9250s get hotter than anything else, presumably because they are self-heating somewhat, but they also have a 105C absolute max operating that we do not exceed while doing temp calibration.

dagar, 2019-04-09:
Thanks for raising that @jzazbert. Let's get a fresh issue going to make sure there's no problem before v1.9.0 goes out.