PX4/PX4-Autopilot #5970: FW auto land switch yaw to wheel controller

dagar, 2016-12-05:
Large yaw actuation during FW landing flare.

http://logs.uaventure.com/view/XDxR5X5pVEjYNULpojSQdB

[image]

The main problem is the switch to the wheel controller here - https://github.com/PX4/Firmware/blob/master/src/modules/fw_pos_control_l1/fw_pos_control_l1_main.cpp#L1622
@AndreasAntener @tumbili I think one of you implemented the wheel controller and landing?
How should we solve this? I thought about hiding it behind runway takeoff enable, but there are plenty of little planes with steerable landing gear that you might want to hand launch. Should we try to detect touchdown?

The next question is why did the yaw setpoint drop 0 (_target_bearing) instead of maintain yaw?

Reported here - http://discuss.px4.io/t/issues-with-landing-rudder-on-flying-wing/1863/13

dagar, 2016-12-10:
@AndreasAntener @tumbili @LorenzMeier thoughts on the following
 - enable/disable the wheel controller based on RWTO_TKOFF
 - enable/disable the wheel controller based on a new param
 - try to detect touchdown before enabling (probably more work with testing)

I'll also figure out what happened with the yaw setpoint at flare, which could be the only thing to actually fix here.

dagar, 2016-12-11:
@Kjkinney reported the issue and is willing to help test solutions.

AndreasAntener, 2016-12-11:
I had a quick look in the source but couldn't find an obvious reason for the strange yaw setpoint.

dagar, 2017-01-26:
Reproducible in SITL.

dagar, 2017-02-06:
The problem here is that it's possible to skip over heading hold, but then in flare the uninitialized _target_bearing is used as the yaw setpoint while the wheel controller is enabled.

[image]

We can fix this by setting _target_bearing properly, but does heading hold as horizontal distance to the landing waypoint make sense? It seems like progress along the glide slope or simply right before flare would be more appropriate.

PR #6536 (2017-02-12): FW auto land force heading hold at flare
 - fixes #5970

@AndreasAntener or @tumbili would you mind reviewing?

It's possible to not hit the heading hold section of the current auto land, then in flare the uninitialized _target_bearing is used for the yaw setpoint as the wheel controller is enabled.

This PR forces heading hold at flare.
closed 2017-02-12 by LorenzMeier