PX4/PX4-Autopilot #5976: X-Plane simulation not working on v1.5.1rc3

vvranjek, 2016-12-05:
Hi,

I wanted to give the latest rc3 a try today and I was unable to get it running with X-Plane 10 simulator. It connects with no problems, but when I want to start the mission/take-off nothing happens. Just like the simulator doesn't receive any data. It was working on v1.5.0, just before I tried the stock v1.5.1rc3. Airframe of choice: HILstar.

Can someone confirm this please?

Cheers

vvranjek, 2016-12-06:
Just found out, I am not the only one. http://discuss.px4.io/t/xplain-10-and-1-5-1rc1/1794

vvranjek, 2016-12-06:
Found the solution! qGCS didn't understand the updated mavlink messages properly. Updated qGCS to the latest daily build and it works just fine.

dagar, 2016-12-11:
Thanks for positing the solution.

closed 2016-12-11 by dagar