PX4/PX4-Autopilot #5978: Intel Aero: Inspect mag interference

LorenzMeier, 2016-12-05:
It has a big issue with the mag. @jgoppert That explains why LPE is struggling. @lfelipe @lucasdemarchi I checked and we should be using the mag in the GPS. It should not be so influenced by current.

[image]

LorenzMeier, 2016-12-05:
This might be enough to fix it: https://github.com/PX4/Firmware/pull/5979

The proper fix is to use the right priorities in all drivers. But this is the safer, quick option for the release.

lucasdemarchi, 2016-12-05:
I didn't know start order matters. Isn't there a parameter to select primary compass?

LorenzMeier, 2016-12-05:
It happens on publication (ORB_PRIO_HIGH). I bet that this is inconsistent in one of the drivers, leading to the result.

PR #5991 (2016-12-07): hmc5883: fix MAGIOCGEXTERNAL ioctl for non fmuv1 boards
This makes the onboard check consistent with hmc5883_bus_option
initialization. If the current bus is the onboard bus, return !external.

It fixes the onboard mag (HMC5883) for AeroFC. Its priority is now 100,
previously it was 255, so that if an external mag is attached, it will be
preferred.

[image]

Fixes #5978
closed 2016-12-07 by LorenzMeier