PX4/PX4-Autopilot #5984: pos_sp_triplet updates continuously

Stifael, 2016-12-06:
Currently the navigator sends position triplets continuously during mission, even when the triplets stay the exact same as during the previous iteration. That leads to unnecessary sending of messages. In my opinion, the update/publish should occur once the triplet has changed.

AndreasAntener, 2016-12-07:
As long as there are altitude updates (FOH in FW) and heading updates (in MC) there will be many updates with slightly changing contents. If the concern is that this causes to much unchanged data copying we would need to reconsider the setpoint triplet interface.

Stifael, 2016-12-07:
ok, I see that the triplet's position setpoint message contains yaw. 
The way I understand it is as follow: 
In not mission mode, the position controller takes care of the yaw setpoint. In mission mode, the navigator takes care of the yaw setpoint, unless the the position setpoint type is "follow target", where again the update occurs in the position controller.

What is the reason for not having the position controller being always in charge of setting the yaw setpoint? In my understanding the navigator could just send out the waypoint reference position triplet and the yaw waypoint settpoint once, and the position controller tries then to achieve that target. If the reference changes or the position controller achieved the target, new setpoints could be sent out.

AndreasAntener, 2016-12-07:
IMHO the right approach would be to only generate the setpoint (which includes yaw) outside the position controllers:
- mission: navigator
- manual: stick mapper (doesn't exist yet, planned for ages though ;))
- follow target: should be a navigator mode (like mission) and not in the position controllers

This separates the responsibilities and allows the setpoint to work over multiple position controllers (for VTOL).

Stifael, 2016-12-08:
I completely agree that the responsibility has to be clearly seperatrd, but currently that is not really the case (yaw setpoint is set from navigator and position controller, depending on the mode). I also see a problem when trying to implement a mission feature where the vehicle has to follow a given track with having the body x-axis always pointing along the track. In this case, the yaw setpoint should be sent by the position controller since he has the infornation to do so. 
Anyway, what I still dont get is why it is necessary to send the triplets continuously during normal mission mode, since the waypoint positions are not changing until the vehicle is close to the current setpoint (yaw is not even considered by the position controller  in normal mission mode). Maybe Im still missing something?

AndreasAntener, 2016-12-08:
> I completely agree that the responsibility has to be clearly seperatrd, but currently that is not really the case (yaw setpoint is set from navigator and position controller, depending on the mode). 

My point is, if we change something, it should go into the right direction. What should be changed is the way manual mode is handled.

> I also see a problem when trying to implement a mission feature where the vehicle has to follow a given track with having the body x-axis always pointing along the track. In this case, the yaw setpoint should be sent by the position controller since he has the infornation to do so.

I was interpreting your reply wrong. The follow target mode is correctly implemented as navigator mode. The yaw setpoint (or yaw rate) is therefore set properly outside the position controller, the `control_auto` code merely copies that.

Stifael, 2016-12-10:
Ok, but then I still dont really understand why the triplet has to be sent continuously, since the position controller just uses the three poses of the triplet. Why is it not possible to send the three points once they have changed (this would require some sort of acknowledge message from the position controller that the new triplet has received)?

Stifael, 2016-12-10:
At least for the mc

dagar, 2016-12-11:
I agree on choosing and working in the right architectural direction, but right now isn't the majority of this FOH which causes a pos_sp update even if the altitude doesn't change?

Require a small alt diff here?
https://github.com/PX4/Firmware/blob/master/src/modules/navigator/mission.cpp#L1020

dagar, 2017-01-16:
So the question is where should the yaw be set? Given that we already send the prev, current, and next position setpoints to the position controller, why not let it determine yaw?

What's the advantage navigator doing it?

Same question for FOH.

closed 2018-02-21 by PX4BuildBot