PX4/PX4-Autopilot #5985: Gazebo optical flow simulation: no position hold

ChristophTobler, 2016-12-06:
I just tested the gazebo optical flow simulation (make posix gazebo_iris_opt_flow) and I could not get a position hold. Altitude hold works and I can see the distance topics in QGC but not the flow topics...

ChristophTobler, 2016-12-09:
I have tested the simulation again with the newest master. Now I get the flow information. Altitude hold again works fine but as soon as I enable position hold the drone crashes.
@jgoppert The output of the flow looks fine. I guess it is something with the lpe? Or is it just wrongly tuned?

jgoppert, 2016-12-09:
@ChristophTobler you need https://github.com/PX4/Firmware/pull/5958. Some of the tests are failing because of size constraints for older boards, but it works. Might have to make a build without ekf2 and just lpe on it.

ChristophTobler, 2016-12-12:
Ok, works with https://github.com/PX4/Firmware/pull/5958. I'll leave the issue open until merged.

ChristophTobler, 2016-12-22:
@jgoppert I saw that https://github.com/PX4/Firmware/pull/5958 has been merged into master. Now is it possible to switch to position hold but the quad dangles heavily. Does it need tuning?

jgoppert, 2016-12-23:
@ChristophTobler For sitl pos hold (make posix_sitl_default gazebo_iris_opt_flow, commander takeoff) there is still a bit of a wobble, could be improved. In real life it works well. Please stay tuned to the PR: https://github.com/PX4/Firmware/pull/6045. I hope to surpass the optical flow performance of LPE with IEKF.