PX4/PX4-Autopilot #5986: build fails if a .msg file is added

CarlOlsson, 2016-12-06:
(This is maybe intended, just close here in that case)
If I add a file to the msg folder (e.g. called test_topic.msg) but not to the CMakeLists.txt the build fails with the error message below. Should the build system ignore a file which is not in the make file?

```
[ 45%] Built target systemcmds__esc_calib
Scanning dependencies of target systemcmds__top
/home/carl/Wingtra/Standard_PX4/Firmware/build_px4fmu-v4_default/src/systemcmds/topic_listener/topic_listener.cpp:74:36: fatal error: uORB/topics/test_topic.h: No such file or directory
 #include <uORB/topics/test_topic.h>
                                    ^
compilation terminated.
make[3]: *** [src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/topic_listener.cpp.obj] Error 1
make[2]: *** [src/systemcmds/topic_listener/CMakeFiles/systemcmds__topic_listener.dir/all] Error 2
make[2]: *** Waiting for unfinished jobs....
[ 45%] Linking C static library libsystemcmds__pwm.a
Building CXX object src/drivers/bmi160/CMakeFiles/drivers__bmi160.dir/bmi160_main.cpp.obj
[ 45%] Building C object src/systemcmds/top/CMakeFiles/systemcmds__top.dir/top.c.obj
[ 45%] Built target systemcmds__pwm
Linking C static library libsystemcmds__top.a
[ 45%] Built target systemcmds__top
Linking CXX static library libdrivers__bmi160.a
[ 45%] Built target drivers__bmi160
make[1]: *** [all] Error 2
make: *** [px4fmu-v4_default] Error 2
```

tumbili, 2016-12-06:
@CarlOlsson you probably have to clean in order to generate the message headers.

bkueng, 2016-12-06:
Yes it's a known issue: the `topic_listener` build command (Tools/generate_listener.py) goes through all .msg files instead of using the CMakeLists.txt.

dagar, 2017-04-11:
Relevant - https://github.com/PX4/Firmware/pull/6609

dagar, 2017-04-12:
Will be resolved by #6609.

closed 2017-04-12 by dagar