PX4/PX4-Autopilot #5989: PWM_AUX_DISARMED not working

tops4u, 2016-12-06:
In current 1.5.1 RC3 I set PWM_AUX_DISARMED to 900. After Power-On is done I get a PWM of roughly 1500 on AUX-1 Pin. After arming the RC Signal is passed through. 

MIN_AUX is set to 900 to, MAX_AUX is set to 2000. I assume this is a bug.

LorenzMeier, 2016-12-06:
Indeed, there was a setup flag missing. Can you try this one?
https://github.com/PX4/Firmware/pull/5990

tops4u, 2016-12-06:
@LorenzMeier Thanks, I'll try tomorrow.

tops4u, 2016-12-07:
@LorenzMeier I don't think this made it in 1.5.1 Final? At least this is not fixed there. Is it in the current Master?

LorenzMeier, 2016-12-07:
I never intended to put it into 1.5.1. You need to test this very branch.

tops4u, 2016-12-07:
@LorenzMeier Currently I don't have a running Build infrastructure up an running. I assume there is no prebuilt package I can download of this branch (at least I did not figure out on GitHub how to do that).

LorenzMeier, 2016-12-07:
What autopilot HW are you using?

tops4u, 2016-12-07:
AUAV-X2 (FMUv2 I think)

LorenzMeier, 2016-12-07:
For Pixracer:
[nuttx-px4fmu-v4-default.px4.zip](https://github.com/PX4/Firmware/files/637798/nuttx-px4fmu-v4-default.px4.zip)

For Pixhawk / AUAV-X2:
[nuttx-px4fmu-v2-default.px4.zip](https://github.com/PX4/Firmware/files/637799/nuttx-px4fmu-v2-default.px4.zip)

tops4u, 2016-12-08:
I'll check later today. Thanks for the builds.

LorenzMeier, 2016-12-09:
Would be great to have your testing on this.

tops4u, 2016-12-09:
@LorenzMeier Apologies I haven't had time yesterday to test as I'm no professional Pilot (work & family). I should find some time today to test it.

LorenzMeier, 2016-12-09:
Great, let me know how you go.

tops4u, 2016-12-09:
@LorenzMeier sorry this does not work. Got a Red-Blue Blinking State. The console showed this: 

> INFO  [init] Mixer: /etc/mixers/hexa_x.main.mix on /dev/pwm_output0 
> INFO  [init] Mixer: /etc/mixers/pass.aux.mix on /dev/pwm_output1 
> ERROR [pwm] PARAM 'PWM_AUX_DISARMED
> ' NAME NOT FOUND

Is there an additional Line Break?

LorenzMeier, 2016-12-13:
Fixed on master.

closed 2016-12-13 by LorenzMeier