PX4/PX4-Autopilot #5995: Barometer altitude hold not working in 1.5.1rc3

rolandash, 2016-12-07:
When trying to test an indoor flight using baro only for altitude hold, the copter just slowly descend down to the ground can can not hold the altitude. No GPS is connected. 

This issue has been seen both in MindPX v2.8 and Pixhawk, running 1.5.1rc3. This issue seems to be independent from which estimator chosen, either LPE or EKF2.

julianoes, 2016-12-08:
Can you share a log please? Thanks.

rolandash, 2017-01-06:
@julianoes 

Sorry for the late response, finally got some time to get the log:

http://logs.uaventure.com/view/Y8B9b5bjFfytqctL3Tth9k

This log is generated on Pixhawk/1.5.3/EKF2. seems the problem still there.

dagar, 2017-04-28:
Any update on this situation?

rolandash, 2017-05-03:
@dagar As 1.5.1rc3 is just an transition version, i did not have much chance to do more flight tests on it. 
However this issue is not observed any more on successor versions. Although i personally still have the feeling that baro altitude estimation does not give me the same confidence as in early 1.4.x versions. Will need more flights to find out.

This issue can be closed.

closed 2017-05-03 by rolandash