PX4/PX4-Autopilot #5997: Intel Aero defconfig is defining CONFIG_ARCH_BOARD_PX4FMU_V4 not CONFIG_ARCH_BOARD_AEROFC_V1

davids5, 2016-12-07:
@lucasdemarchi 

In merging master into nuttx_v3 I noticed that the defconfig for aerofc-v1 is using CONFIG_ARCH_BOARD_PX4FMU_V4 and the CONFIG_ARCH_BOARD is wrong also. 

Once I fixed this it led to an error where the PX4FMU_DEVICE_PATH was not defined.

This indicates to me there may be build inconsistencies based one a command line -DCONFIG_ARCH_BOARD_AEROFC_V1 and what is in the config.h (CONFIG_ARCH_BOARD_PX4FMU_V4)

Furthermore aerofc-v1 added #ifdef CONFIG_ARCH_BOARD_AEROFC_V1 and should not be modifying "PX4 system files" fmu.cpp and tap.cpp 

@lucasdemarchi - please fix this on master and discuss herein the problems you were trying to solve with the ifdefs so we can choose how to clean up the "PX4 system files".

Also note the USB info as defined on nuttx_v3:nuttx-configs/aerofc-v1/nsh/defconfig (if there even is one) - Schematics would help me to see if the defconfig is correct.

lucasdemarchi, 2016-12-07:
> Also note the USB info as defined on nuttx_v3:nuttx-configs/aerofc-v1/nsh/defconfig (if there even is one) - Schematics would help me to see if the defconfig is correct.

there isn't any USB.

> Furthermore aerofc-v1 added #ifdef CONFIG_ARCH_BOARD_AEROFC_V1 and should not be modifying "PX4 system files" fmu.cpp and tap.cpp
> @lucasdemarchi - please fix this on master and discuss herein the [...]

for tap I think I already replied to you there:
https://github.com/PX4/Firmware/pull/5893#discussion_r89103404

for fmu:

The gpio part is really fragile. Any other module could do an ioctl() leading the system to act on a gpio it shouldn't. This avoided tracing back to the caller. Fixing it properly may involve change the functions called, e.g. `gpio_set_function()` to account for cases in which `_ngpio == 0`.

davids5, 2016-12-08:
@lucasdemarchi, fyi @LorenzMeier 

Thank you. 

I will remove the USB from the nuttx_v3 config for aerocf-v1 on the next upstream nuttx update this week.

For fmu what about changing the 
```
#define BOARD_FMU_GPIO_TAB {}
```
then remove all the ifdefs in fmu.cpp and change the type of _ngpio and the for's i vars to int?

Would you please try that and test it and do a PR against master for it.

For the tap_esc:

 If the HW you are using does not conform to the tap_esc spec. We have 3 options:

1) Fix the HW.

2) Change the #ifdef to be more a component oriented definition as apposed to board specific. 
I.E. HAVE_NO_ESC_CONFIG_RESP then change the ifdef in tap_esc to ```#if !defined(HAVE_NO_ESC_CONFIG_RESP)``` wrapping the code the aerofc can not tolerate and light HAVE_NO_ESC_CONFIG_RESP in the areofc-v1's board_config.h
 
3) Clone the tap_esc file set and name it aerofc_esc. Then change it to support areofc HW and swap the tap_esc driver out for the aerofc_esc in the nuttx_aerofc-v1_default.cmake

Option 2 & 3 have value when others can use them. 3 adds a minor maintenance overhead but lowers the coupling effect. After all when binary compatibility is not required module inclusion  in the cmake is a far clearer than a forest of ifdefs.

lucasdemarchi, 2016-12-10:
> If the HW you are using does not conform to the tap_esc spec. We have 3 options:

what's the tap esc spec? I've never seen such spec.

davids5, 2016-12-10:
@lucasdemarchi - my bad! It was a bad choice of word on my part.  "Uses the code as written" would best describe it,

PR #6104 (2016-12-20): aerofc: small fixes
Fixes #5997
closed 2016-12-20 by LorenzMeier