PX4/PX4-Autopilot #5998: ESC Array data (esc_report) not available in px4log files from sdlog2_dump.py

dbarge, 2016-12-08:
Hello,

There seems to be some array data missing in the .px4log files.
I noticed this looking at .csv files generated with sdlog2_dump.py.
So, I'm not sure if the data is missing in the .px4log or if there is a problem with the conversion to .csv.

For example, in a 4 ESC setup (with ESC addresses 100, 101, 102, 103) I only get data for one ESC (103) in the (csv) log.

>./sdlog2_dump.py  file.px4log -m 'ESC'

Gives:

ESC_count,ESC_nESC,ESC_Conn,ESC_N,ESC_Ver,ESC_Adr,ESC_Volt,ESC_Amp,ESC_RPM,ESC_Temp,ESC_SetP,ESC_SetPRAW
1859,4,4,3,103,0,14.5,0.60009765625,1614,306.0,13.0,1

The fields ESC_Ver, ESC_Adr, ESC_Volt, ESC_Amp, ESC_RPM, ESC_Temp, ESC_SetP, ESC_SetPRAW should be arrays.
They correspond to the following fields (from msg/esc_report.msg):

uint16 esc_version                  # Version of current ESC - if supported
uint16 esc_address                  # Address of current ESC (in most cases 1-8 / must be set by driver)
float32 esc_voltage                 # Voltage measured from current ESC [V] - if supported
float32 esc_current                 # Current measured from current ESC [A] - if supported
int32 esc_rpm                       # Motor RPM, negative for reverse rotation [RPM] - if supported
float32 esc_temperature                 # Temperature measured from current ESC [degC] - if supported
float32 esc_setpoint                    # setpoint of current ESC
uint16 esc_setpoint_raw                 # setpoint of current ESC (Value sent to ESC)

3 fields seem to be missing:

uint16 esc_state                    # State of ESC - depend on Vendor
uint8 esc_vendor                    # Vendor of current ESC
uint32 esc_errorcount                   # Number of reported errors by ESC - if supported

Also, ESC_Ver and ESC_Adr seem to be swapped.

dbarge, 2016-12-09:
I can look into this more...
It would help if someone could tell me where in the code to look to see how the uorb data is related to what is written to the logs.
For example, How does esc_address in msg/esc_report.msg make it to ESC_Adr in the px4log?

Thanks

dagar, 2016-12-09:
For sdlog2 it happens here - https://github.com/PX4/Firmware/blob/master/src/modules/sdlog2/sdlog2.c#L2068

For the new logger it can log the orb message (msg/esc_report.msg) directly.

dbarge, 2016-12-10:
Thanks,

What's the new logger?
Can you point me to documentation or where it lives?

Thanks!

dagar, 2016-12-10:
http://dev.px4.io/advanced-logging.html

closed 2017-04-03 by mhkabir