PX4/PX4-Autopilot #6003: RTL operation: Vehicle holds position

LorenzMeier, 2016-12-08:
http://logs.uaventure.com/view/qH2GpFUzaeVXcQtXkbHb8B

Had an issue during the flight, flew in Stabilize first and no issues during the flight, Lost RC Signal and entered RTL, vehicle did not want to descend, switched to Position and Altitude and didn't want to descend while throttle was held down, had to descend with Stabilize at this point, issue happened 2 times with the same battery.

LorenzMeier, 2016-12-08:
I think this is the wrong log or a pilot confusion. The vehicle never went into RTL (NAV STATE 5):

[image]

julianoes, 2016-12-08:
I have to agree with @LorenzMeier.

LorenzMeier, 2016-12-09:
This is apparently the real log:
http://logs.uaventure.com/view/qH2GpFUzaeVXcQtXkbHb8B

LorenzMeier, 2016-12-09:
Still makes no sense. There is no sequence where the altitude is constant and the pilot input is not constant. Wondering if the RC just stopped providing any outputs.

tops4u, 2016-12-09:
Maybe this log will help you: http://logs.uaventure.com/view/Y9qXUQzcXUbiKHWdiFJpdD

This is on 1.5.1 (RC 1 or 3).

Did two RTL Tries and stopped them before the Vehicle could descent to the ground. I did never experience any controll Issues. Note that I have a Spectrum RC System, so my Receiver will set the Mode Switch to RTL PWM when Signal is Lost, as Spektrum does not output any RSSI Information.

pkocmoud, 2016-12-19:
What I saw on a recent flight was I triggered RTL via RC channel. The aircraft returned and landed.  Switched out of RTL and resumed flying. Every later attempt to RTL would result in the vehicle returning but not attempting to descend to land again. This continued until power cycled.

closed 2018-02-21 by PX4BuildBot