PX4/PX4-Autopilot #6004: New Logger not functioning on Pixhawk mini

santiago3dr, 2016-12-08:
running 1.5.1-stable on pixhawk mini
able to arm and fly but no log recorded on sd card
flight: https://youtu.be/CDstI6IEuc0

screenshots of parameter and sd content below

[image]
[image]
[image]

bkueng, 2016-12-09:
Did you flash the px4fmu-v2 target? The `logger` module is not built there because of flash restrictions.

kd0aij, 2016-12-09:
The fmu-v3 target should work for Pixhawk1 compatibles that have rev3
silicon, and it does have the new logger.

closed 2016-12-09 by LorenzMeier