PX4/PX4-Autopilot #6016: No Position Information below EKF2 using RTK GPS

meteor0530, 2016-12-10:
Hi
when I am using a RTK below EKF2, QGC displays "RTK 3D lock".But the x,y informantion in LOCAL_POSITION_NED in mavlink inspector are both zero,and it cant be switched to POSCTRL.
However, when I replace EKF2 with INAV, everything goes well, my vehicle can detect the GPS and can be switched to POSCTRL.
Do I need to tune some parameters?
Need some helps,Thanks!

LorenzMeier, 2016-12-10:
Can you share a log file from an open field with v1.5.2 or master? We generally only offer support for the last stable release. The last releases did not include INAV btw. You might be using quite dated software.

meteor0530, 2016-12-12:
@LorenzMeier Thanks for your reply
I think my version is v1.4.4, do I need to share a log with v1.4.4?

mhkabir, 2016-12-12:
@meteor0530 Please update to v1.5.2 and then share a log.

meteor0530, 2016-12-16:
@mhkabir 
Here is the log: [http://logs.uaventure.com/view/jgmeKKBTRrVWfQhvG8CzM7#GPS_Uncertainty_PLOT](url)

meteor0530, 2016-12-20:
@LorenzMeier  @mhkabir 
Have you seen my log? whats wrong with my px4?

mhkabir, 2017-04-03:
Closing as stale. This is most likely fixed as EKF2 has undergone significant testing since then. Please reopen if required.

closed 2017-04-03 by mhkabir