PX4/PX4-Autopilot #6017: Rework attitude tuning UI

LorenzMeier, 2016-12-10:
@anton-matosov It would be great if you could work with @DonLakeFlyer on a tuning UI for multicopters that is more user-friendly (new or different sliders).

anton-matosov, 2016-12-10:
I already have implemented PID tuning UI for MC that is mobile friendly.
I will post screenshots shortly.
It might be a good starting point.

anton-matosov, 2016-12-10:
[image]

anton-matosov, 2016-12-10:
[image]
[image]

dagar, 2016-12-11:
Plots? https://github.com/mavlink/qgroundcontrol/issues/2878#issuecomment-188401859

dagar, 2016-12-11:
QGC issue - https://github.com/mavlink/qgroundcontrol/issues/3911

anton-matosov, 2016-12-11:
Plotting can be pretty easily done with QML http://doc.qt.io/qt-5/qtcharts-qmloscilloscope-example.html
I need clear definitions of what should be plotted, as I have never really used plots for tuning.
We should also consider how plotting and tuning can fit for the mobile (phone/phablet/tablet) as it is much more convenient to tune from device then from laptop especially over Bluetooth

anton-matosov, 2016-12-11:
BTW, is anybody working on BLE support for datalink? This seems to be the only other option except WIFI to connect on the iOS device

If not I will give it a try

LorenzMeier, 2016-12-11:
1) BLE:
@sirPerna Any news on BLE?

@anton-matosov Here are the things that you will want to plot against each other:

2) PLOTTING: It should be selectable as pairs: roll vs. roll setpoint, roll rate vs roll rate setpoint. You get the attitude from this message:
http://mavlink.org/messages/common#ATTITUDE_QUATERNION
and setpoint:
http://mavlink.org/messages/common#ATTITUDE_TARGET

Here is how you convert from Quaternion to roll / pitch / yaw:
https://github.com/mavlink/c_library_v2/blob/master/mavlink_conversions.h#L93-L106

sirPerna, 2016-12-11:
@anton-matosov As I know nobody started yet. If you want to give it a try I'm happy to support you. TBS Crossfire has BLE support and MAVLink is already built-in.

anton-matosov, 2016-12-14:
I am going to use simple [HM-10 module](http://fab.cba.mit.edu/classes/863.15/doc/tutorials/programming/bluetooth/bluetooth40_en.pdf) and connect it to telemetry port as I do with HM-05

dagar, 2017-01-02:
Any progress? I'm really looking forward to not digging through the param tree.

anton-matosov, 2017-01-03:
Not really. I was busy during past couple of weeks. I hope to submit first PR with basic tuning by end of the week.

anton-matosov, 2017-01-21:
first iteration https://github.com/mavlink/qgroundcontrol/pull/4458

dagar, 2017-04-11:
@anton-matosov were you still working on this? The last screenshot I saw looked like a huge improvement.

anton-matosov, 2017-04-27:
Sorry @dagar, I didn't have enough time to finish this and do not expect to work on this in foreseeable feature.
Latest code is here https://github.com/anton-matosov/qgroundcontrol/tree/PID-Tuning-v2
It is stable, but some filters are not very obvious to the user and need more tweaking (that is the missing part I planned to implement for v2)
You can pull this code from my branch and work with @DonLakeFlyer to get it in.

dagar, 2017-04-27:
No problem, I still think the current state is a drastic improvement. Thanks!

dagar, 2017-04-27:
New PR - https://github.com/mavlink/qgroundcontrol/pull/5070

closed 2018-02-21 by PX4BuildBot