PX4/PX4-Autopilot #6021: CCache broken on Semaphore

LorenzMeier, 2016-12-10:
@dagar I don't think its completely disabled (but I think we should completely disable it) can you have a look?

```
[12/474] Building NuttX for px4io-v1
FAILED: cd /home/runner/Firmware/build_px4fmu-v1_default && /bin/mkdir -p /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX && rsync -a --delete --exclude=.git ../NuttX/ px4io-v1/NuttX/ && /usr/bin/make --no-print-directory -C/home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX/nuttx -r --quiet distclean && /bin/cp -r /home/runner/Firmware/nuttx-configs/PX4_Warnings.mk /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX/nuttx/ && /bin/cp -r /home/runner/Firmware/nuttx-configs/px4io-v1 /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX/nuttx/configs && cd /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX/nuttx/tools && ./configure.sh px4io-v1/nsh && cd .. && /usr/bin/make --no-print-directory --quiet -C /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX/nuttx -j4 -r CONFIG_ARCH_BOARD=px4io-v1 export > nuttx_build.log && /bin/cp -r /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1/NuttX/nuttx/nuttx-export.zip /home/runner/Firmware/build_px4fmu-v1_default/px4io-v1.export
ccache: error: /root/.ccache/ccache.conf: No such file or directory
make[3]: *** [mkconfig] Error 1
make[2]: *** [tools/mkconfig] Error 2
make[2]: *** Waiting for unfinished jobs....
```

dagar, 2016-12-10:
Where are you seeing this? I also added CCACHE_DISABLE to the environment, but don't see any recent failures.

LorenzMeier, 2016-12-10:
Here. I did it now set in the Makefile, which finally fixed it:
https://github.com/PX4/Firmware/pull/6022

Commit:
https://github.com/PX4/Firmware/pull/6022/commits/3bf27c70d27708a78d388b05a617350e1c4dfcb9

dagar, 2016-12-10:
We're mainly just running out of space on semaphoreci, not hitting actual ccache problems.

Are you sure you want to disable ccache everywhere? It's going to make travis-ci builds much slower. Developers also have it on locally.

LorenzMeier, 2016-12-10:
If you have a more selective method I'm happy to consider but right now CI is totally out of whack and killing productivity. I can't let it dangle and I also have only limited time to fix it.

dagar, 2016-12-10:
For semaphore it's disabled in the docker run now.

    docker run --rm -v `pwd`:`pwd`:rw -e CI=true -e CCACHE_DISABLE=1 -w=`pwd` --user=root -it px4io/px4-dev-base:2016-10-20 /bin/bash -c "make check"

Could we build bootloaders in a separate thread to get around the space issue?

LorenzMeier, 2016-12-10:
It will not scale. We need to limit the CCACHE size or disable it *and* delete build artefacts after building. Fundamentally the CI system setup needs to build one target and then build the next one. Otherwise we will soon run into the same issue.

Splitting bootloaders can still make sense in terms of runtime and would be a welcome addition.

LorenzMeier, 2016-12-10:
Fixed with your env variable change, thanks!

closed 2016-12-10 by LorenzMeier